rmk-rs/rmk #7: BLE support for nRF52832

lonesometraveler, 2024-03-13:
Thank you for developing and maintaining this incredible project.

I've noticed that the BLE support is currently limited to nRF52840. Are there any plans to include support for nRF52832? If not, I'm willing to take on the task of adding support for nRF52832. I believe this will broaden the project's accessibility.

HaoboGu, 2024-03-13:
Yeah, supporting other chips of nRF is on the future list, but I don't have a board for nRF52832/nRF52810/etc. It would be great if you could do this :D

lonesometraveler, 2024-03-13:
Great. I have nRF52832 boards. I will work on it.

HaoboGu, 2024-03-14:
I just made a minor modification of feature gates, end users would add only `nrf528xx_ble` feature, and `ble` feature will be added then.

HaoboGu, 2024-03-14:
Could you please update [rmk-template](https://github.com/HaoboGu/rmk-template) as well, making nRF52832 ble feature available for more users?

lonesometraveler, 2024-03-14:
I am not familiar with `cargo-generate`. But I'm open to learning and giving it a try.

LXF-YZP, 2024-03-19:
> I just made a minor modification of feature gates, end users would add only 我只是对功能门进行了小幅修改，最终用户只会添加`nrf528xx_ble`feature, and feature，以及`ble`feature will be added then. 然后将添加功能。

> Could you please update [rmk-template](https://github.com/HaoboGu/rmk-template) as well, making nRF52832 ble feature available for more users?

Hello, can 52840 be edited online in vial? Whether it's wired or wireless?

HaoboGu, 2024-03-19:
@LXF-YZP can only wired version of 52840 use vial for now. I'm doing a storage refactoring, which makes it possible to switch between usb/ble, then you can use vial after switching to usb mode. 

I also planned to test whether vial could be used in wireless mode, I don't know whether it's possible. PRs welcome!

HaoboGu, 2024-03-23:
nRF52832 is now supported, closing this

closed 2024-03-23 by HaoboGu