rmk-rs/rmk #22: No connected probes were found

LXF-YZP, 2024-03-20:
After I have placed rmk clone locally, and entered the nrf52840_ble directory, I run the cargo run --release command and get the following error:

`    cargo run --release 
    Finished release [optimized + debuginfo] target(s) in 0.14s
     Running `probe-rs run --chip nRF52840_xxAA target/thumbv7em-none-eabi/release/rmk-nrf52840`
Error: No connected probes were found.`

Use jlink to connect to my pad, connect four interfaces swd, swc, gnd, vcc, my pad uses e73 module, e73 module uses 52840.
My pad used to run zmk firmware.

rust version ：1.76.0
system：macos12.7.4

HaoboGu, 2024-03-20:
Could you attach the result of `probe-rs list`?

LXF-YZP, 2024-03-20:
> Could you attach the result of `probe-rs list`? I suspect that probe-rs doesn't recognize your debugging probe
Here is the output

` probe-rs list                                                                                          
The following debug probes were found:
[0]: J-Link (J-Link) (VID: 1366, PID: 0101, Serial: 000000123456, JLink)`

HaoboGu, 2024-03-20:
Could you make sure that jlink is connected to nrf52840's swd debugging port? 

How was you flash zmk firmware to your module?

LXF-YZP, 2024-03-20:
> Could you attach the result of `probe-rs list`? I suspect that probe-rs doesn't recognize your debugging probe

> Could you make sure that jlink is connected to nrf52840's swd debugging port?
> 
> How was you flash zmk firmware to your module?

The zmk firmware is in uf2 format and only needs to be dragged and dropped

LXF-YZP, 2024-03-20:
> Could you make sure that jlink is connected to nrf52840's swd debugging port?
> 
> How did you flash zmk firmware to your module?

reerror

`cargo run --release                                                                                                        
    Finished release [optimized + debuginfo] target(s) in 0.13s
     Running `probe-rs run --chip nrf52840_xxAA target/thumbv7em-none-eabihf/release/rmk-demo`
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An ARM specific error occurred.
    1: An ARM specific error occurred.
    2: Failed to connect to the debug port. Please check the debug cable and target power. If SWD multi-drop is used, ensure the correct TARGETSEL value is used.`

HaoboGu, 2024-03-20:
that should be a debug probe issue. You can try to convert rmk-nrf52840_ble firmware to bin then to uf2 and then drag&drop the uf2

LXF-YZP, 2024-03-20:
> that should be a debug probe issue. You can try to convert rmk-nrf52840_ble firmware to bin then to uf2 and then drag&drop the uf2

Can you tell me which file you want to convert? Because there's only one folder in the command.thanks

HaoboGu, 2024-03-20:
By default Rust generates elf file, which is `target/thumbv7em-none-eabi/release/rmk-nrf52840_ble`.

You can use [cargo binutils](https://github.com/rust-embedded/cargo-binutils) to generate .bin file using `cargo objcopy --release -- -O binary app.bin`

Then convert `app.bin` to uf2 using microsoft's uf2conv tool: https://github.com/Microsoft/uf2/blob/master/utils/uf2conv.py

LXF-YZP, 2024-03-20:
> There is another way to check: you can download nrf's ble stack using
> 
> ```
> probe-rs erase --chip nrf52840_xxAA
> probe-rs download --verify --format hex --chip nRF52840_xxAA s140_nrf52_7.3.0_softdevice.hex
> ```
> 
> If you still get connection error, that should be a debug probe or probe-rs issue.

Still having a connection problem

LXF-YZP, 2024-03-20:
> By default Rust generates elf file, which is `target/thumbv7em-none-eabi/release/rmk-nrf52840_ble`.
>
> You can use [cargo binutils](https://github.com/rust-embedded/cargo-binutils) to generate .bin [...]

There are two similar files in the release directory, rmk-nrf52840 and rmk-nrf52840.d。
One of the two files is 1.8 megabytes and the other is more than 400 bytes。
Both file sizes are confusing. If it's over 400 bytes, it's too small.

HaoboGu, 2024-03-20:
`rmk-nrf52840` is the correct one

HaoboGu, 2024-03-20:
> Still having a connection problem

Yeah, that's clear. it's not a RMK issue. You should double check whether you could connect nrf52840 with your jlink. Using other tools like pyocd or openocd could do this. If you can connect using other tools, this maybe a probe-rs bug, otherwise it's probably your hardware's issue. 

I'm using CmsisDap so I have no idea about why jlink doesn't work.

LXF-YZP, 2024-03-20:
> Yeah, that's clear. it's not a RMK issue. You should double check whether you could connect nrf52840 with your jlink. Using other tools like pyocd or openocd could do this. If you can connect using [...]

Thanks for your reply.

LXF-YZP, 2024-03-20:
> `rmk-nrf52840` is the correct one

I tried to convert it, but the file felt a little too large at 1.8M.
Will uf2 firmware be generated when the subsequent version is updated? rp2040 and nrf52840 brush uf2 format files are very convenient, this is just my habit of using qmk, and zmk.

HaoboGu, 2024-03-20:
> I tried to convert it, but the file felt a little too large at 1.8M.

Shouldn't be that large. I tried before, iirc it was about 100KB. 

> Will uf2 firmware be generated when the subsequent version is updated?

You can open another issue for this feature, I'll do it if I have time.

LXF-YZP, 2024-03-20:
> > I tried to convert it, but the file felt a little too large at 1.8M.
> 
> Shouldn't be that large. I tried before, iirc it was about 100KB.
> 
> > Will uf2 firmware be generated when the subsequent version is updated?
> 
> You can open another issue for this feature, I'll do it if I have time.

Thanks. I'll try

HaoboGu, 2024-03-25:
@LXF-YZP I've added some docs about converting RMK's firmware to UF2 format and flashing via USB, you can check it out: https://github.com/HaoboGu/rmk/tree/main/boards/nrf52840_ble