rmk-rs/rmk #27: nRF52832 Runtime error with version 0.1.9

lonesometraveler, 2024-03-25:
Since the nrf-softdevice flash module's `capacity` function [returns an incorrect value for nRF52832](https://github.com/embassy-rs/nrf-softdevice/issues/246), a nRF52832 BLE program always fails at runtime [here](https://github.com/HaoboGu/rmk/blob/e83b261226523b3f8478fac3f0c6afc2bb0f89e8/rmk/src/storage.rs#L300):

```rust
impl<F: AsyncNorFlash> Storage<F> {
    pub async fn new<const ROW: usize, const COL: usize, const NUM_LAYER: usize>(
        flash: F,
        keymap: &[[[KeyAction; COL]; ROW]; NUM_LAYER],
    ) -> Self {
        let storage_range = (flash.capacity() - 2 * F::ERASE_SIZE) as u32..flash.capacity() as u32;
        let mut storage = Self {
            flash,
            storage_range,
        };

        // Check whether keymap and configs have been storaged in flash
        if !storage.check_enable::<ROW, COL, NUM_LAYER>().await {
            // Initialize storage from keymap and config
            if let Err(e) = storage.initialize_storage_with_config(keymap).await {
                error!("Initialize storage error: {}", e)
            }
        }

        storage
    }
```

HaoboGu, 2024-03-26:
A configuration could be added to set the flash range for storage. Current implementation requires at least 2 sectors available. Some constraints could be added as well. I can do it this week.

HaoboGu, 2024-03-26:
I've pushed a commit that enables setting storage info, you can check it out.

lonesometraveler, 2024-03-26:
Confirmed, it works.

closed 2024-03-26 by lonesometraveler