rmk-rs/rmk #29: add touchpad support

wallamide, 2024-03-30:
there's a rust driver for the [azoteq tps-65](https://mm.digikey.com/Volume0/opasdata/d220001/medias/docus/2228/TPS43,_TPS65_Datasheet_Web.pdf) available [here](https://github.com/rwalkr/eskarp/blob/main/firmware/device/src/touchpad.rs). it would be nice for this to get ported to rmk. i will be trying to get this working on my current keyboard build with n!n and a tps43, but i'm making slow progress on my project so i'd like to put this out there for others to try if they'd like.

HaoboGu, 2024-04-11:
@wallamide I might have some time in the next few months, I'll look into it. The question is, where can I get a azoteq tps-65 for testing? Touchpad support is a feature that I definitely want

EzioL, 2024-05-10:
> @wallamide I might have some time in the next few months, I'll look into it. The question is, where can I get a azoteq tps-65 for testing? Touchpad support is a feature that I definitely want

贸泽或者1688找南京赛泰

whitelynx, 2024-11-11:
> @wallamide I might have some time in the next few months, I'll look into it. The question is, where can I get a azoteq tps-65 for testing? Touchpad support is a feature that I definitely want

It looks like it's available from a couple of places, depending on your preferences:
- [Mouser](https://eu.mouser.com/ProductDetail/Azoteq/TPS65-201A-S?qs=pfd5qewlna5Lh8O0E8DcUQ%3D%3D)
- [DigiKey](https://www.digikey.com/en/products/detail/azoteq-pty-ltd/TPS65-201A-S/7164942)

It looks like this touchpad in particular is an I2C part, but I know that there's others which interface via SPI, serial, PS/2, or USB. (I believe older ThinkPad models used PS/2 for their touchpad, since they also used that for the TrackPoint module)

HaoboGu, 2024-11-11:
> > @wallamide I might have some time in the next few months, I'll look into it. The question is, where can I get a azoteq tps-65 for testing? Touchpad support is a feature that I definitely want
> 
> @HaoboGu It looks like it's available from a couple of places, depending on your preferences:
> 
> * [Mouser](https://eu.mouser.com/ProductDetail/Azoteq/TPS65-201A-S?qs=pfd5qewlna5Lh8O0E8DcUQ%3D%3D)
> * [DigiKey](https://www.digikey.com/en/products/detail/azoteq-pty-ltd/TPS65-201A-S/7164942)
> 
> It looks like this touchpad in particular is an I2C part, but I know that there's others which interface via SPI, serial, PS/2, or USB. (I believe older ThinkPad models used PS/2 for their touchpad, since they also used that for the TrackPoint module)

Yes, I have bought one, but haven't got time starting on it 🥲

I'm thinking about a extendable input device system, which should be easy to add more devices, whether it speaks i2c/spi/serial/etc. This is hard without alloc. 

And, some of them are like mouses -- they continuously send events, which may cause key_event_channel congestion.

whitelynx, 2024-11-17:
I love the extendable input device system idea! It would make it easier for me to add support for the TrackPoint.

As far as the event channel congestion, maybe we can do something to filter out unnecessary events, but if any pointing device is actively being moved, it's going to generate a lot of event traffic. I know that there's QMK-based devices with various pointing devices which work fine, and I would assume that RMK should be able to achieve similar performance, so I don't think it will be a huge issue in practice, but we'll have to wait and see how performance looks once we've implemented a device.

HaoboGu, 2026-08-29:
closing this because iqs5xx support has been merged

closed 2026-08-29 by HaoboGu