rmk-rs/rmk #33: How do I solve the row2col problem

LXF-YZP, 2024-04-08:
My pcb board is row2col. Currently, it seems that there is no way to manually change col2row to row2col in the rmk project

HaoboGu, 2024-04-08:
There're some bugs for row2col impl, I'll fix it soon

LXF-YZP, 2024-04-08:
> There're some bugs for row2col impl, I'll fix it soon

thanks

HaoboGu, 2024-04-08:
I've pushed a fix to main branch, you can try it by disabling default features to use row2col in your `Cargo.toml`

```rust
rmk = { version = "0.1.12", git = "https://github.com/HaoboGu/rmk", default-features = false, features = ["nrf52840_ble"]}
```

That's because RMK enables `col2row` feature **by default**. So if you want to use row2col, you have to disable default features and then add every feature gate that you're using to `features` field. This could be optimized though.

LXF-YZP, 2024-04-08:
> I've pushed a fix to main branch, you can try it by disabling default features to use row2col in your 我已经将修复程序推送到主分支，您可以通过禁用默认功能来尝试在您的`Cargo.toml`
> 
> ```rust
> rmk = { version = "0.1.12", git = "https://github.com/HaoboGu/rmk", default-features = false, features = ["nrf52840_ble"]}
> ```
> 
> That's because RMK enables 那是因为 RMK 使`col2row`feature 特征**by default 默认情况下**. So if you want to use row2col, you have to disable default features and then add every feature gate that you're using to . 因此，如果要使用 row2col，则必须禁用默认功能，然后添加您正在使用的每个功能门`features`field. This could be optimized though. 田。不过，这可以优化。

I appreciate your message

HaoboGu, 2024-04-08:
Closing this. If you speak Chinese, you can join rust embedded community (which is a QQ group 763148381) and ask me :D

LXF-YZP, 2024-04-08:
谢谢，我会加入。

closed 2024-04-08 by HaoboGu