rmk-rs/rmk #39: Macropad (RP2040) 'matrix-less' switch support?

erdii, 2024-05-27:
Heya :wave: 

I was wondering: does this firmware support the "matrix-less" inputs on the Adafruit Macropad [1]?
The macropad has 12 keyboard switches, each connected to it's own GPIO pin.

Cheers,
Josh

[1] https://www.adafruit.com/product/5128

HaoboGu, 2024-05-27:
Hi!

Direct pin is not supported right now, but adding this feature isn't hard imo. If you really want it, I can add it in next weeks, or you can do a PR and I'm happy to review it :D

erdii, 2024-05-29:
Ah that's a bummer.

I haven't written any rust actively and my experience only revolves around gargabe-collected languages so far, so I'm not sure if my rust-foo is good enough to contribute without getting overwhelmed by the steep learning curves of rust, cross-compiling rust for embedded and embassy but I'll dabble a few more hours to try and find out. :)

Thanks for creating this project and being a maintainer of it and I'd like to acknowledge that I don't have any entitlement to ask you for adding this feature (in the next weeks or at all). 

That said, If you'd like to share any pointers on where to start, I'd happily take them. I'd also happily download a version of rmk that includes the feature.

Cheers! :)

HaoboGu, 2024-05-30:
You can follow the first two sections of [user guide](https://haobogu.github.io/rmk/guide_overview.html) to setup Rust and RMK environment. 

If you're not familiar with Rust nor Rust embedded, the official [Rust book](https://doc.rust-lang.org/book/) and [the Rust embedded book](https://docs.rust-embedded.org/book/) are the good start points of your Rust journey.

Happy coding! :D

HaoboGu, 2024-06-13:
I just did some investigation. Current `matrix.rs` implementation is limited to traditional keyboard matrix, but it can be abstracted easily to a trait which is used in `keyboard.rs`:

```rust
pub(crate) trait KeyboardMatrix {
    fn new() -> Self;
    fn get_key_state(&mut self, row: usize, col: usize) -> KeyState;
    fn update_timer(&mut self, row: usize, col: usize);
    async fn scan(&mut self);
    #[cfg(feature = "async_matrix")]
    async fn wait_for_key(&mut self); 
}
```

Then the direct pin matrix can be implemented with very little effort. I currently just doesn't have enough time to implement it. But if anyone want to do, I'd happy to give suggestions & guidance about it.

davidqge, 2024-08-25:
I don't have hardware to try. But I think a one row matrix would work. Give it an unused pin as row.

decaday, 2024-10-29:
#116 has been merged.
I have implemented a DirectPinMatrix, including the DirectPinMatrix struct, modules, configuration file options, procedural macro entry functions, examples, and documentation.

Examples are:
https://github.com/HaoboGu/rmk/tree/main/examples/use_config/rp2040_direct_pin
https://github.com/HaoboGu/rmk/tree/main/examples/use_rust/rp2040_direct_pin

HaoboGu, 2024-10-29:
I'm closing this since #116 has been merged. @erdii would you like to have a test?

closed 2024-10-29 by HaoboGu