rmk-rs/rmk #43: The rp2040 example doesn't work on my rpi-pico board.

IotaHydrae, 2024-06-01:
linux mint 21.2
Offical raspberrypi pico core board.

I tried
```
cargo build
warning: unused import: `Blocking`
  --> src/main.rs:16:20
   |
16 |     flash::{Async, Blocking, Flash},
   |                    ^^^^^^^^
   |
   = note: `#[warn(unused_imports)]` on by default

warning: `rmk-rp2040` (bin "rmk-rp2040") generated 1 warning (run `cargo fix --bin "rmk-rp2040"` to apply 1 suggestion)
    Finished dev [optimized + debuginfo] target(s) in 12.49s
```
and
```
cargo run
warning: unused import: `Blocking`
  --> src/main.rs:16:20
   |
16 |     flash::{Async, Blocking, Flash},
   |                    ^^^^^^^^
   |
   = note: `#[warn(unused_imports)]` on by default

warning: `rmk-rp2040` (bin "rmk-rp2040") generated 1 warning (run `cargo fix --bin "rmk-rp2040"` to apply 1 suggestion)
    Finished dev [optimized + debuginfo] target(s) in 0.04s
     Running `probe-rs run --chip RP2040 target/thumbv6m-none-eabi/debug/rmk-rp2040`
      Erasing ✔ [00:00:01] [####] 116.00 KiB/116.00 KiB @ 79.42 KiB/s (eta 0s )
  Programming ✔ [00:00:02] [####] 116.00 KiB/116.00 KiB @ 46.39 KiB/s (eta 0s )    Finished in 3.991s
```
But the Pico board doesn't seem to have any response, I mean after connecting the USB cable to the computer.

Did I miss any steps? Thanks.

HaoboGu, 2024-06-01:
That seems correct. Do you see any other log output after running `cargo run`?

I just tested my pico board, all good:

[image]

Or you can try `lsusb -t` to check whether your host recognizes your board

HaoboGu, 2024-06-01:
Add this line to your `Cargo.toml` if the code panics with a task-arena size error:

```diff
embassy-executor = { version = "0.5", features = [
    "defmt",
    "arch-cortex-m",
    "executor-thread",
    "integrated-timers",
+   "task-arena-size-8192",
] }
```

IotaHydrae, 2024-06-01:
Nothing. Also, I have tried `cargo run --release`. It's just waiting there, same with `cargo run`.
[image]

With adding `task-arena-size-8192` to `Cargo.toml`, I got a USB device
```
Bus 001 Device 026: ID 4c4b:4643 Haobo RMK Keyboard
```
Seems okay, so did the log output.
```
❯ cargo run
warning: unused import: `Blocking`
  --> src/main.rs:16:20
   |
16 |     flash::{Async, Blocking, Flash},
   |                    ^^^^^^^^
   |
   = note: `#[warn(unused_imports)]` on by default

warning: `rmk-rp2040` (bin "rmk-rp2040") generated 1 warning (run `cargo fix --bin "rmk-rp2040"` to apply 1 suggestion)
    Finished dev [optimized + debuginfo] target(s) in 0.04s
     Running `probe-rs run --chip RP2040 target/thumbv6m-none-eabi/debug/rmk-rp2040`
      Erasing ✔ [00:00:01] [##################] 116.00 KiB/116.00 KiB @ 80.57 KiB/s (eta 0s )
  Programming ✔ [00:00:02] [##################] 116.00 KiB/116.00 KiB @ 47.20 KiB/s (eta 0s )    Finished in 3.927s
INFO  RMK start!
└─ rmk_rp2040::____embassy_main_task::{async_fn#0} @ src/main.rs:36
INFO  USB: config_descriptor used: 122
└─ embassy_usb::builder::{impl#1}::build @ /home/iotah/.cargo/registry/src/mirrors.ustc.edu.cn-61ef6e0cd06fb9b8/embassy-usb-0.2.0/src/fmt.rs:143
INFO  USB: bos_descriptor used: 12
└─ embassy_usb::builder::{impl#1}::build @ /home/iotah/.cargo/registry/src/mirrors.ustc.edu.cn-61ef6e0cd06fb9b8/embassy-usb-0.2.0/src/fmt.rs:143
INFO  USB: msos_descriptor used: 0
└─ embassy_usb::builder::{impl#1}::build @ /home/iotah/.cargo/registry/src/mirrors.ustc.edu.cn-61ef6e0cd06fb9b8/embassy-usb-0.2.0/src/fmt.rs:143
INFO  USB: control_buf size: 128
└─ embassy_usb::builder::{impl#1}::build @ /home/iotah/.cargo/registry/src/mirrors.ustc.edu.cn-61ef6e0cd06fb9b8/embassy-usb-0.2.0/src/fmt.rs:143
INFO  Device enabled
└─ rmk::usb::{impl#3}::enabled @ /home/iotah/embeddedboys/rmk/rmk/src/usb.rs:189
INFO  Device suspended, the Vbus current limit is 500µA (or 2.5mA for high-power devices with remote wakeup enabled).
└─ rmk::usb::{impl#3}::suspended @ /home/iotah/embeddedboys/rmk/rmk/src/usb.rs:219
```

Thanks again for your help.

closed 2024-06-01 by IotaHydrae

HaoboGu, 2024-06-01:
okay! I've pushed a commit adding this to all examples. Thanks for the feedback!