rmk-rs/rmk #49: Modifier Keys not working in combination with normal keys on MacOS

flamestro, 2024-07-22:
Hey,

I have tried to create a key combination like LSpace(KC_A). 
However, this gets registered as "a" not as "A".

Same for LGui(KC_Space) which gets registered as space and not as LGui Space (which would open the spotlight search on macos).

I checked the keyboard.rs, however, I was not yet able to find why this happens. I tried on two different RP2040's.

I set the macros / keymap via vial.rocks

Maybe anyone has an idea why this happens.

HaoboGu, 2024-07-23:
I should be a bug, I'm fixing it.

HaoboGu, 2024-07-23:
https://github.com/HaoboGu/rmk/pull/50/  should fix this.

flamestro, 2024-07-23:
It does, thanks for taking a look. 

However, it is not a fix for all use cases.

It works for a single modifier + key assignment but not within a macro.

This is probably because `process_action_macro` does not use the method you fixed. 

I am still trying to figure out how it could be integrated. Maybe you have an idea?

HaoboGu, 2024-07-24:
Ah, yeah, macro has similar logic for texts, thanks for reporting it, I'll fix it soon.

Current macro implementation has some limitations: it supports only basic keys, since complex key actions like `LT(k)` need async recursive implementation of `process_action_macro`, which requires everything to be static. I haven't found a better solution for it:(

HaoboGu, 2024-07-24:
#51 fixes keyboard macro. I can achieve this, is it what you need?

https://github.com/user-attachments/assets/3a150d2c-9578-4911-9749-3583c970fe9d

flamestro, 2024-07-24:
Yes, thanks Haobo!
Thats a nice workaround.

HaoboGu, 2024-07-24:
Cool! I'll close this issue. Thanks for your feedback! 

If you have other problems using this library, feel free to open another one :D

closed 2024-07-24 by HaoboGu