rmk-rs/rmk #55: [Feature Request] Chords and Layers and Modifiers combined

w-jablonski, 2024-08-15:
Thank you for this promising project!

Then you get people like me who come and start Feature Requests like: I need 777 features that are present in QMK, but done better! So please forgive.

Scenario: I wanna have numbers under left hand like so: press a `Button_1`+`{z,x,c,...}` simultaneously to type `{1,2,3,...}` so like Chords in QMK. Easy. But then, if typing longer numbers, why repeat these chords with each digit? No, I wanna keep holding `Button_1` down while I release the `{z,x,c,...}` and then tap the subsequent `{z,x,c,...}` keys to type subsequent digits. So in summary, pressing such a Chord activates a layer which remains active until the `Button_1` is released.

If we implement config for features like Chords, Layers separately (like in QMK), then there might be no elegant way to mix them, like in above scenario. So ideally, these would be declared in a one, unified way which allows for any possible scenario, including Chords, Modifiers, Layers, One-shot Action, Tap-or-hold, Multiple Tap (known as Tap Dance in QMK).

I could try design the elegant config, but not the implementation since I don't know Rust. Or maybe this has been done before by another project?

HaoboGu, 2024-08-16:
RMK doesn't have Chord right now, I'm still considering how these features(Chord, Layers, Tap/Hold, One-Shot, etc) can be implemented in an ergonomic way. Obviously, they have something common but are totally different. If I understand you correct, this feature is somehow like a `partial layer switch`: `Button_1` actually activates another layer, but only for the specific keys.

w-jablonski, 2024-08-16:
Thanks. I am not quite sure about "partial layer switch". I see it simply as activating another layer, even if a small one consisting of only 10 keys.

HaoboGu, 2024-08-16:
Yeah that's what I mean.

HaoboGu, 2025-01-14:
> @HaoboGu, there is already a very powerful keyboard customization application called [Kanata](https://github.com/jtroo/kanata) that is also written in Rust. Would there be any way to make their configuration scheme work with this project?

Kanata is written for desktop, not for the embedded. It's better to have a converter which converts Kanata scheme to RMK's config. I'm quite open to it if someone wants to take over

tib888, 2025-04-13:
> Thanks. I am not quite sure about "partial layer switch". I see it simply as activating another layer, even if a small one consisting of only 10 keys.

So is this a layer where only that 10 keys work, and if anything else is pressed, then immediately turns off that layer, and accepts the pressed "other" key also?

w-jablonski, 2025-04-14:
> then immediately turns off that layer

No, that would be a bonus feature but I wouldn't need it. Not what I meant.

HaoboGu, 2025-07-15:
I think this can be done by combining layer and with_modifier keys

closed 2025-07-15 by HaoboGu