rmk-rs/rmk #57: Crash in ssmarshal upon press key

davidqge, 2024-08-23:
I changed esp32c3 to esp32c6. Had to increase stack size significantly in order to fix the crash before ble starts.
Now the BLE can connect, but crashes when press any key.

code is here: https://github.com/davidqge/rmk/tree/esp32c6_crash

`I (2608) esp32_nimble::server::ble_server: mtu update event; conn_handle=1 mtu=256
thread 'main' panicked at /home/david-embedded/.cargo/registry/src/index.crates.io-6f17d22bba15001f/ssmarshal-1.0.0/src/lib.rs:125:5:
8 <=? 4
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

abort() was called at PC 0x42036897 on core 0
0x42036897 - panic_abort::__rust_start_panic::abort
    at /home/david-embedded/.rustup/toolchains/esp/lib/rustlib/src/rust/library/panic_abort/src/lib.rs:49
Core  0 register dump:
MEPC    : 0x408014d0  RA      : 0x4080a8ba  SP      : 0x4082bcc0  GP      : 0x40810720
0x408014d0 - panic_abort
    at /home/david-embedded/esp-idf-template/.embuild/espressif/esp-idf/v5.2.2/components/esp_system/panic.c:466
0x4080a8ba - __ubsan_include
    at /home/david-embedded/esp-idf-template/.embuild/espressif/esp-idf/v5.2.2/components/esp_system/ubsan.c:313
0x40810720 - coex_schm_ble_mesh_standby_bt_a2dp_paused_wifi_connecting
    at ??:??
TP      : 0x407feec0  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x4082bcfc  S1      : 0x4082bcfc  A0      : 0x4082bcfc  A1      : 0x4082bcde
A2      : 0x00000000  A3      : 0x4082bd29  A4      : 0x00000001  A5      : 0x40814000
0x40814000 - ble_sm_proc_mem
    at ??:??
A6      : 0x00000000  A7      : 0x76757473  S2      : 0x4082bce0  S3      : 0x4082be14
S4      : 0x00000001  S5      : 0x00000000  S6      : 0x00000001  S7      : 0x420b0e80
0x420b0e80 - anon.58211478a7d4bb73acbde57ab30debcb.3.llvm.17918663921895458077
    at ??:??
S8      : 0x00000003  S9      : 0x00000001  S10     : 0x4082ead8  S11     : 0x4082eadc
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
0x40800001 - _vector_table
    at ??:??
MHARTID : 0x00000000

Stack memory:
4082bcc0: 0x4082bcd0 0x00000003 0x4082bcfc 0x4080edec 0x420c1210 0x00000001 0x00000004 0x00000030
0x4080edec - __assert_func
    at /home/david-embedded/esp-idf-template/.embuild/espressif/esp-idf/v5.2.2/components/newlib/assert.c:34
4082bce0: 0x33303234 0x37393836 0x0325dd00 0x4081200c 0x4082bce0 0x40812028 0x4082bcdc 0x726f6261
4082bd00: 0x20292874 0x20736177 0x6c6c6163 0x61206465 0x43502074 0x34783020 0x36333032 0x20373938
4082bd20: 0x63206e6f 0x2065726f 0x00000030 0x40840000 0x420c28b8 0x00000000 0x40812a94 0x4203689a
0x40812a94 - std::panicking::HOOK
    at ??:??
0x4203689a - object::endian::U16Bytes<E>::get
    at /home/david-embedded/.cargo/registry/src/index.crates.io-6f17d22bba15001f/object-0.36.0/src/endian.rs:636
4082bd40: 0x4082bd38 0x4082bd28 0x4082bd2c 0x4082bd27 0x00000000 0x00000000 0x00000000 0x4202c062
0x4202c062 - core::ptr::write
    at /home/david-embedded/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/ptr/mod.rs:1534
4082bd60: 0x420c28b8 0x00000000 0x408417e0 0x4202cf40 0x420c28b8 0x00000000 0x40812a94 0x420224f4
0x4202cf40 - std::thread::local::LocalKey<T>::try_with
    at /home/david-embedded/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/thread/local.rs:282
0x40812a94 - std::panicking::HOOK
    at ??:??
0x420224f4 - core::fmt::Arguments::new_const
    at /home/david-embedded/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/fmt/mod.rs:343
4082bd80: 0x010e99b4 0x00000000 0x40810388 0x40802a8a 0x4082be14 0x420c1228 0x4082be70 0x420b0e80
0x40810388 - s_stub_table
    at ??:??
0x40802a8a - esp_timer_impl_set_alarm_id
    at /home/david-embedded/esp-idf-template/.embuild/espressif/esp-idf/v5.2.2/components/esp_timer/src/esp_timer_impl_systimer.c:84
0x420b0e80 - anon.58211478a7d4bb73acbde57ab30debcb.3.llvm.17918663921895458077
    at ??:??
4082bda0: 0x00030001 0x00000000 0x40841d40 0x408026f6 0x0107a69a 0x00000000 0x40841d40 0x40802860
0x408026f6 - timer_insert
    at /home/david-embedded/esp-idf-template/.embuild/espressif/esp-idf/v5.2.2/components/esp_timer/src/esp_timer.c:340
0x40802860 - esp_timer_start_once
    at /home/david-embedded/esp-idf-template/.embuild/espressif/esp-idf/v5.2.2/components/esp_timer/src/esp_timer.c:230
4082bdc0: 0x00000000 0x00000000 0xa5a5a5a5 0x00000000 0x80000006 0x00000000 0x40841c80 0x80000006
4082bde0: 0x4082be44 0x4082ed64 0x00000001 0x4082ed74 0x4082ed70 0x4082ead8 0x4082ed5c 0x4082ed94
4082be00: 0x40830cc0 0x4082be70 0x4082be34 0x42024d6a 0x80000006 0x00000008 0x408418c0 0x00000007
0x42024d6a - std::panicking::begin_panic_handler::{{closure}}
    at /home/david-embedded/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/panicking.rs:658
4082be20: 0x4082be70 0x4082eae8 0x4082ed94 0x42024c24 0x40830cc0 0x4082be70 0x4082be4c 0x420b0e80
0x42024c24 - std::sys_common::backtrace::output_filename
    at /home/david-embedded/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std/src/sys_common/backtrace.rs:182
0x420b0e80 - anon.58211478a7d4bb73acbde57ab30debcb.3.llvm.17918663921895458077
    at ??:??
4082be40: 0x00000000 0x80000006 0x00000000 0x00000001 0x420cd3b0 0x4082be70 0x420b0e80 0x42000001
0x420cd3b0 - anon.eb3a7f1d3ba0a8f1a939908ae7979056.5.llvm.2562215316079177712
    at ??:??
0x420b0e80 - anon.58211478a7d4bb73acbde57ab30debcb.3.llvm.17918663921895458077
    at ??:??
4082be60: 0x40812544 0x4082ed74 0x00000020 0x00000008 0x420b0e0c 0x00000002 0x4082be88 0x00000002
0x40812544 - rmk::ble::esp::initialize_esp_ble_keyboard_with_config_and_run::{{closure}}::keyboard_channel
    at ??:??
0x420b0e0c - anon.58211478a7d4bb73acbde57ab30debcb.2.llvm.17918663921895458077
    at ??:??
4082be80: 0x00000000 0x40836660 0x4082be6c 0x42047b04 0x420b0514 0x42047b04 0x4082ed30 0x420023f4
0x42047b04 - <u32 as core::fmt::num::DisplayInt>::to_u32
    at /home/david-embedded/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/fmt/num.rs:44
0x420b0514 - anon.33830a95427e32f1388f8325fe794489.48.llvm.6508362364345514770
    at ??:??
0x42047b04 - <u32 as core::fmt::num::DisplayInt>::to_u32
    at /home/david-embedded/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/fmt/num.rs:44
0x420023f4 - rmk::ble::esp::server::<impl rmk::hid::HidWriterWrapper for alloc::sync::Arc<esp32_nimble::utilities::mutex::Mutex<esp32_nimble::server::ble_characteristic::BLECharacteristic>>>::write_serialize::{{closure}}
    at /home/david-embedded/esp-idf-template/rmk/rmk/src/ble/esp/server.rs:35
4082bea0: 0x00000000 0x4082ea6c 0x4082eae0 0x00000001 0x40830cc0 0x40830cc0 0x4082edb8 0x00000000
4082bec0: 0x00000007 0x40820000 0x4082edb8 0x40830cb8 0x4082cf30 0x4082ea6c 0x4082eae0 0x00000001
4082bee0: 0x408309e0 0x4082eae8 0x40830cc0 0x420030b0 0x00000000 0x4082f9e0 0x4082edb8 0x42005022
0x420030b0 - <embassy_futures::select::Select3<A,B,C> as core::future::future::Future>::poll
    at ??:??
0x42005022 - rmk::ble::esp::initialize_esp_ble_keyboard_with_config_and_run::{{closure}}
    at /home/david-embedded/esp-idf-template/rmk/rmk/src/ble/esp.rs:107
4082bf00: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x4082ebf0 0x4082ebe0 0x4082ebd0 0x420b2ed0
4082bf20: 0x00000019 0x00000010 0x0000000f 0x0000000e 0x4082ed94 0x4082eb70 0x420b2728 0x4082eb38
4082bf40: 0x40830a20 0x40830a10 0x40830a00 0x00000018 0x420b0518 0x40830c35 0x40830a28 0x40830c65
0x420b0518 - anon.1fa986d0522e96aa06299dcaab5fe17e.192.llvm.11678645180993410721
    at ??:??
4082bf60: 0x408309fc 0x40830a2c 0x4082ed5c 0x408309e8 0x4082da04 0x000009c8 0x40812540 0x4082d99c
4082bf80: 0x4082d948 0x4082d9a3 0x4082e96c 0x4082e90c 0x4082da0c 0x4082d998 0x420b270c 0x4082ea78
0x420b270c - anon.eb91ea43833730bbd2d6b31f28c614c8.5.llvm.5682508570986425722
    at ??:??
4082bfa0: 0x4082d9b0 0x4082cf50 0x4082cf10 0x4082f9e0 0x4082ed30 0x4082eaf8 0x4082ea74 0x4082ea70
4082bfc0: 0x4082edb8 0x4082cf4c 0x4082cf48 0x4082cf38 0x4082f9dc 0x4082eae8 0x01010101 0x4082d9d0
4082bfe0: 0x40830c68 0x40830c98 0x4082cf80 0x00000f00 0x4082eae4 0x408308e0 0x0025f06e 0x00000000
4082c000: 0xa5a5a500 0xa5a5a5a5 0xa5a5a5a5 0x40837178 0x00000002 0x0000001e 0x00020000 0x001f0000
4082c020: 0x00000000 0x00000002 0x00000020 0x00020000 0x00210000 0x00000000 0x00000002 0x00000022
4082c040: 0x00020000 0x00230000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4082c060: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
4082c080: 0x00000000 0x00000000 0x00000002 0x0000002b 0x00020000 0x00140000 0x00000000 0x00000002
4082c0a0: 0x0000001a 0x00020000 0x00080000 0x00000000 0x00000002 0x00000015 0x00020000 0x00170000

`

HaoboGu, 2024-08-24:
Thanks for reporting it. Could you provide full backtrace by enabling RUST_BACKTRACE=1?

davidqge, 2024-08-28:
Changed write_serialize to write fixed the problem.
Created a pull request along with several other fixes to make it work on esp32c6.

closed 2024-08-28 by davidqge