rmk-rs/rmk #62: Refactor storage interface as plain key value store

davidqge, 2024-09-04:
Due to this bug https://github.com/esp-rs/espflash/issues/676, can't use embedded storage on esp-idf.

Currently, the storage.rs mix many rmk types with sequential_storage. Make it difficult to utilize the existing nvs key value storage included in esp-idf.

Propose to refactor storage into plain key value store, and keep type specific code in each own class.

HaoboGu, 2024-09-05:
esp-storage is for esp-hal(no_std), not esp-idf-hal. For storage support for esp, you can check out this issue: https://github.com/esp-rs/esp-idf-svc/issues/405 .

HaoboGu, 2024-09-05:
The storage implementation for esp32 has been deprioritized because esp-idf doesn't have usb ported right now, esp32c3/c6 also don't have general usb peripheral, so even though we have storage implemented for esp32, we still can't use vial.

I'm open for a storage implementation which is better suit for RMK's usage, which should support very limited flash, wear-leveling feature, etc. That requires lots of efforts, but now I don't have enough time for that.

davidqge, 2024-09-05:
I saw this project, so bought this esp to use vial. 

I didn't know Vial didn't work over BLE. Anyway, I changed Vial, now can connected over BLE.

Then rmk freeze because of storage queue full. So need to get storage working.

HaoboGu, 2024-09-06:
> I didn't know Vial didn't work over BLE. Anyway, I changed Vial, now can connected over BLE.

If vial can connect over BLE, it's reasonable to add storage support to esp. I would suggest to implement `embedded-storage` for `nvs` in `esp-idf-svc`, then you can use it as RMK's storage seamlessly.

davidqge, 2024-09-13:
embedded-storage is NorFlash trait.  nvs is higher level key-value store,  like sequential-storage.
I implemented embedded-storage for idf partition, can keep use sequential-storage now.

But startup loading keymap config take 8s, because each fetch requires scan whole data.
I am thinking iterate record at lower level, thus only need to go through data one time.

closed 2024-09-19 by davidqge