rmk-rs/rmk #69: Tracking issue for Encoder support

tanekere, 2024-09-17:
Hi 👋,
I wanted to build a project like this one when i found yours. I am not new to rust or embedded programming but am new to embedded programming in Rust. Would this be a good first issue for me?

HaoboGu, 2024-09-17:
Yes, thanks for raising the issue! You can post anything about encoder support here:D

tanekere, 2024-09-23:
Took a look into this today and looks like most rotary encoders rely on pull-up inputs. But as far as i can tell the rmk-macro crate can only support pull down inputs as of right now. So i guess that will be the first task. 
Also, the macros just refer to them as 'input_pin' (eg.`convert_gpio_str_to_input_pin`) so should i change names to pd_input or keep it like this and just add functions for pull up?

azzamsa, 2024-11-08:
Wow, `sm_td` was mentioned. I'm always afraid of leaving `sm_td` behind, as I relied on this after I migrated from Miryoku (but I keep the exact layout). As an avid Rust programmer myself, I can't wait to try `rmk` HRM for my next board. Hopefully, this week!

HaoboGu, 2025-04-01:
I've already added rotary encoder support, it's available by using Rust code only. I'll add the config soon. Please checkout the example [here](https://github.com/HaoboGu/rmk/blob/main/examples/use_rust/nrf52840_ble/src/main.rs).