rmk-rs/rmk #87: [Feature Request] SPI 74HC595 shift register as part of matrix scanning

ramonimbao, 2024-10-06:
ZMK has support for [74HC595 shift registers as parts of its matrix scanning](https://zmk.dev/docs/development/hardware-integration/shift-registers) as of adding additional GPIO. This works really well for small boards with few GPIO available like the [Seeed Studio XIAO NRF52840](https://www.seeedstudio.com/Seeed-XIAO-BLE-nRF52840-p-5201.html). I have boards here I can test for this. I'd like to be able to contribute but I am unfamiliar with embedded Rust, and my Rust skills are definitely rusty 😅

HaoboGu, 2024-10-06:
I'm happy to see more matrix types are supported :D

You can combine [`embedded_hal::spi`](https://docs.rs/embedded-hal/latest/embedded_hal/spi/index.html) and [`MatrixTrait`](https://github.com/HaoboGu/rmk/blob/main/rmk/src/matrix.rs#L10C1-L15C31) to write a platform diagnostic 74HC595 matrix scanning algorithm. If you have any question about implementing it, just feel free to ask

HaoboGu, 2024-10-06:
If you knew Rust before, learning embedded stuff is not that hard. I think some basic embedded and spi protocol knowledge is all you need to implement it.

ramonimbao, 2024-10-07:
Was searching around for crates to reference and I found this crate: [shiftreg-spi](https://crates.io/crates/shiftreg-spi). It seems like exactly what's needed since it implements the `OutputPin` trait. Does this mean we'll be able to mix it in with a normal matrix alongside other microcontroller GPIO pins? That's exactly how some of my boards are set up.

HaoboGu, 2024-10-07:
> Was searching around for crates to reference and I found this crate: [shiftreg-spi](https://crates.io/crates/shiftreg-spi). It seems like exactly what's needed since it implements the OutputPin [...]

Yes, it seems that you even don't need to change any RMK code. If I understand it correctly, you can just put the `OutputPin` list to the matrix, that should work.

mingti, 2026-04-01:
Any progress on this? I'd love to use this feature in my project if it's already in the main branch.

closed 2026-04-15 by HaoboGu