rmk-rs/rmk #94: `cargo build --release --bin peripheral` does not output hex file

akarsh1995, 2024-10-13:
When I initially downloaded rmk and tried an example it generated hex files and converted to uf2 as well.

I'm not sure what went wrong that `cargo build --release --bin peripheral` stopped generating hex file.

[Asciinema Recording](https://asciinema.org/a/wng0OPcik4EWYHN4sqUtF27mh)

Following is the modification diff
[corne-build-rmk.patch](https://github.com/user-attachments/files/17355967/corne-build-rmk.patch)

akarsh1995, 2024-10-13:
My terminal alacritty had an I/O issue. 
When I tried with macos stock terminal binaries output as expected.

closed 2024-10-13 by akarsh1995

HaoboGu, 2024-10-13:
Hi！cargo build won't generate hex file. If you want to generate uf2, you can follow the instruction here: https://haobogu.github.io/rmk/user_guide/4_compile_and_flash.html#use-uf2-bootloader

HaoboGu, 2024-10-13:
If you want hex only, you should use cargo objcopy, which requires cargo-binutil.

akarsh1995, 2024-10-13:
This command `cargo build --release --bin central` outputs bin right?

HaoboGu, 2024-10-13:
> This command `cargo build --release --bin central` outputs bin right?

it generates ELF file actually

akarsh1995, 2024-10-13:
> > This command `cargo build --release --bin central` outputs bin right?
> 
> it generates ELF file actually

I again, cannot locate elf/bin after running the command `cargo build --release --bin central` in the `examples/use_rust/nrf52840_ble_split`.

### The build process just finishes like so

```
   Compiling embassy-nrf v0.2.0 (/Users/akarshjain/temp/embassy/embassy-nrf)
    Finished `release` profile [optimized + debuginfo] target(s) in 21.00s
```

Am I missing something?

HaoboGu, 2024-10-13:
> I again, cannot locate elf/bin after running the command `cargo build --release --bin central` in the `examples/use_rust/nrf52840_ble_split`.
>
>
>
> ### The build process just finishes like so [...]

the generated ELF is at target/<ARCH>/release. The ELF filename is the bin name set in Cargo.toml, with no extension

akarsh1995, 2024-10-13:
> > 
> > 
> > > > This command `cargo build --release --bin central` outputs bin right?
> > 
> > > 
> > 
> > > it generates ELF file actually
> > 
> > 
> > 
> > 
> > 
> > 
> > 
> > 
> > 
> > 
> > 
> > I again, cannot locate elf/bin after running the command `cargo build --release --bin central` in the `examples/use_rust/nrf52840_ble_split`.
> > 
> > 
> > 
> > ### The build process just finishes like so
> > 
> > 
> > 
> > ```
> > 
> >    Compiling embassy-nrf v0.2.0 (/Users/akarshjain/temp/embassy/embassy-nrf)
> > 
> >     Finished `release` profile [optimized + debuginfo] target(s) in 21.00s
> > 
> > ```
> > 
> > 
> > 
> > 
> > 
> > Am I missing something?
> > 
> > 
> > 
> > 
> > 
> > 
> 
> the generated ELF is at target/YOUR_ARCH/release. The ELF filename is the bin name set in Cargo.toml, with no extension 

Sorry for the noob question but

how do I convert elf to uf2 so that I could flash it via bootloader in my nrf52840?

HaoboGu, 2024-10-13:
ahh, doesn't `cargo make uf2 --release` work?

HaoboGu, 2024-10-13:
It seems that python is not installed at your machine. can you try `python --version` command?

akarsh1995, 2024-10-13:
[image]

python isn't there python3 is

I'll soft link it to python

HaoboGu, 2024-10-13:
You can also change `python` in `Makefile.toml` to use `python3`. I think I'd add more documents about this..