rmk-rs/rmk #98: Create a Discord Channel for RMK Community Development

akarsh1995, 2024-10-14:
RMK has the potential to stand out as a strong alternative to both ZMK and QMK. While ZMK and QMK enjoy substantial community support, I believe there is an opportunity to setup a community around RMK as well.

To support this initiative, I propose creating a dedicated Discord channel where RMK users, developers, and enthusiasts can collaborate, share knowledge, and contribute to the development of RMK.

This channel would help engage the community, provide a space for discussions, and encourage new contributors to get involved, thereby strengthening the RMK ecosystem.

HaoboGu, 2024-10-14:
I do have a discord channel and I put a badge for discord in readme.. But only a few people joined🥲

akarsh1995, 2024-10-14:
Apart from the badge can you also add the hyperlink in the Readme. That way it's easier to search. I searched for the term discord but couldn't find it. I totally missed the badge at the top.

You get the sense

[image]

closed 2024-10-14 by HaoboGu