rmk-rs/rmk #113: Odd behaviour with tap-hold keys

jackbrad1ey, 2024-10-27:
I've been running into some rather odd behaviour with the tap-hold keys. Sometimes my keyboard seems to just get stuck in the layer that it's meant to switch to when holding, it doesn't always revert back to the original layer once I release the key. This also seems to introduce some other wacky behaviour, such as some keys just not working entirely after typing for a little when it gets stuck, and there's a condition where it seems to just continually spam a key since it seems to register it being pressed but never released.
Unsure if this is an issue in the code, or if maybe I'm running into some memory issues? I'm certain this would have been seen by someone else if it was a problem with the code, so just curious if this is a known problem or if there's something else going on here.

HaoboGu, 2024-10-27:
Do all those odd behaviors happen only for tap-hold keys?

HaoboGu, 2024-10-27:
Ah I found a bug here, could you please test whether the PR fixes this?

jackbrad1ey, 2024-10-27:
Just gave it a test, the issue still seems to be present

jackbrad1ey, 2024-10-27:
It seems that the issue appears when I press a key during the 200ms threshold, so for example I press the tap/hold key and then quickly press another key it gets stuck

HaoboGu, 2024-10-28:
thanks for the info! I will take a look today

HaoboGu, 2024-10-28:
Another fix is added. I don't have board at this time.. Will test it later

jackbrad1ey, 2024-10-28:
I just gave it a test, seems to be working flawlessly now, haven't been able to get the bug to come back

closed 2024-10-28 by HaoboGu