rmk-rs/rmk #118: Better HRM(home row mod) support

HaoboGu, 2024-10-30:
Reference:

- QMK tweak for better HRM: https://github.com/stasmarkin/sm_td
- ZMK config with "Timeless HRM": https://github.com/urob/zmk-config

MarkuBu, 2024-11-03:
Looks like an issue for me too because typing fails if I type faster. I need to type slower and more "clear" with home row mods. If I type a key while still releasing the last one it doesn't work.

HaoboGu, 2024-11-03:
which version did you use? could you try latest release?

MarkuBu, 2024-11-03:
I used the template. I assume that it is the latest release, but not sure.

Maybe some details: my keymap in the keymap.toml is empty. Meaning that I just have "_" for each key. I configured everything with Vial.

For HRM I used e.g. LCtrl_T, LGui_T and so on plus the letter.

If I type faster I tend to type "sloppy", meaning that I type a new letter until I didn't fully released the last one. I need to type more careful and release a key before I press the next.

HaoboGu, 2024-11-03:
> For HRM I used e.g. LCtrl_T, LGui_T and so on plus the letter.
> 
> 
> 
> If I type faster I tend to type "sloppy", meaning that I type a new letter until I didn't fully released the last one. I need to type more careful and release a key before I press the next.

what you expect is they should be recognized as "tap", but sometimes when you typing fast, some "hold" actions are triggered since the previous key is not fully released, am I right?

MarkuBu, 2024-11-03:
Correct

HaoboGu, 2024-11-04:
Okay, I got it. The current implementation is too strict about the timing of pressing/releasing tap-hold keys. Let me find a way to solve this.

azzamsa, 2024-11-08:
Wow, `sm_td` was mentioned. I'm always afraid of leaving `sm_td` behind, as I relied on this after I migrated from Miryoku (but I keep the exact layout). As an avid Rust programmer myself, I can't wait to try `rmk` HRM for my next board. Hopefully, this week!

Related:
- https://github.com/stasmarkin/sm_td/issues/28

willpuckett, 2024-11-30:
I'm just discovering RMK and very excited about it, so apologies if I'm chiming in uninformed... ZMK has a 'require-prior-idle' option with it's home row mods that prevents them from triggering if another key has been pressed within a given window... I use that along with 'hold-trigger-key-positions' which restricts the mods to only the opposite half of the keyboard witin the tap window.

HaoboGu, 2024-12-01:
> I'm just discovering RMK and very excited about it, so apologies if I'm chiming in uninformed... ZMK has a 'require-prior-idle' option with it's home row mods that prevents them from triggering if [...]

Thanks for the info! I've implemented something like 'require-prior-idle' and the time can be configured in keyboard.toml. About 'hold-trigger-key-positions', does this config mean that the mods on the other half can be triggered even though it's pressed within the 'prior-idle-time'?

willpuckett, 2024-12-01:
'hold-trigger-key-positions' is part of ZMK's 'positional hold tap'. During the tapping window, a 'hold' can ONLY be triggered by the keys in the supplied list. This assists in preventing accidental triggers when rolling which is especially common on highly ergonomic layouts. It also reduces latency by triggering a tap immediately if the subsequent key press is NOT in the supplied list. 

I imagine that RPI would cover a large number of cases, but with the two of them combined, I have really no issues with accidental holds.

 I'm going to try a RMK build for my split after I finish a small gaming pad I'm doing for a friend now... I'm not finding the RPI feature in the RMK docs I'd be excired to give it a whirl if I can get pointed in the right direction...

HaoboGu, 2024-12-01:
> 'hold-trigger-key-positions' is part of ZMK's 'positional hold tap'. During the tapping window, a 'hold' can ONLY be triggered by the keys in the supplied list. This assists in preventing accidental [...]

yeah, the branch is not merged yet. The document is at: https://github.com/HaoboGu/rmk/blob/feat/hrm/docs/src/keyboard_configuration.md#tap-hold

And, you have to change your branch to 'feat/hrm' to use the HRM code.

willpuckett, 2024-12-05:
@HaoboGu Thanks for rebasing the hrm pr to include direct splits! Will try soon.

I'm a little uncertain how to use in the keymap?

HaoboGu, 2024-12-05:
> I'm a little uncertain how to use in the keymap?

@willpuckett you can follow the [user guide](https://haobogu.github.io/rmk/user_guide/1_guide_overview.html). 

[Here](https://haobogu.github.io/rmk/keyboard_configuration.html#layout) is the documentation for setting keymap, and [here](https://haobogu.github.io/rmk/keyboard_configuration.html#matrix) for setting direct pin matrix

willpuckett, 2024-12-05:
Thanks I've looked through the user guide and the direct pin makes sense. What I'm not sure about is how to call a home row mod in the keymap? Would it be LT(LM(1, LShift), R) for instance, to have a shift hrm on the R key?

HaoboGu, 2024-12-05:
> Thanks I've looked through the user guide and the direct pin makes sense. What I'm not sure about is how to call a home row mod in the keymap? Would it be LT(LM(1, LShift), R) for instance, to have a shift hrm on the R key?

Yes, LT and MT represent LayerTapHold and ModifierTapHold. I think using [vial](https://vial.rocks) to change the keymap is more convenient and intuitive

PR #125 (2024-12-11): Refactor tap/hold processing, add better HRMs support out-of-the-box
Current tap/hold implementation has some issues, see #118, #124. This PR proposes an approach which addresses the issue, as well as HRM config.

The proposed solution is actually a mixture of [urob's zmk-config](https://github.com/urob/zmk-config), [Achordion](https://getreuer.info/posts/keyboards/achordion/index.html) and [sm_td](https://github.com/stasmarkin/sm_td). Many respect to the authors! 

The following is a simple description of new tap/hold processing:
When the keyboard processes tap/hold keys, it checks `prior_idl [... cut]