rmk-rs/rmk #119: Memory issues when adding more layers

jackbrad1ey, 2024-10-31:
Hey apologies for another question, hopefully a quick one though. When testing earlier I know you recommended adding `"task-arena-size-32768"` to my embassy-executor in the `cargo.toml`. Can this task arena size go even higher? Currently when I go past 3 layers nothing works which I assume is again an issue of running out of space in memory, and also for some reason when I press my delete key the keyboard just stops working for whatever reason. I had a hunt through embassy's documentation but couldn't find any reference to `"task-arena-size-32768"` or anything that would let me increase this size further, so just curious if you have any ideas.

HaoboGu, 2024-10-31:
Sure, you can find all available settings [here](https://github.com/embassy-rs/embassy/blob/main/embassy-executor/Cargo.toml#L105).

> and also for some reason when I press my delete key the keyboard just stops working for whatever reason

does this happen only when pressing `delete` key? could you provide more context?

jackbrad1ey, 2024-10-31:
Cheers, I bumped up the size and now extra layers are working fine. As for the delete key, I think that one is a mix of a problem with embassy and my hardware. I designed one specific key on my keyboard (which happens to be the one I've also chosen for Delete) to pull the BOOT pin of my RP2040 low while also acting as a normal keyboard key, so on startup if I have it held down it'll enter USB boot mode.
[image]

Now this is fine as long as the software doesn't care about the state of this pin at runtime, I can confirm on a hardware level this works as my test software acted perfectly fine with this. So either embassy or rmk is monitoring the state of BOOT at runtime and causing some funny business to occur, if you have any clues then some pointers in the right direction would be great, but all good if not as this might be totally unrelated to rmk, I'll have to do a little digging.

HaoboGu, 2024-10-31:
Great, I'm closing this since the memory issue is solved. Feel free to open another issue if you have progress on the boot key issue :P

closed 2024-10-31 by HaoboGu

decaday, 2024-10-31:
The BOOT pin of the RP2040 is different from other MCUs like the STM32. The BOOT pin of the RP2040 is essentially the SPI_CS pin. If there is an incorrect voltage level during operation, the RP2040 will be unable to read the Flash and continue running the program.

jackbrad1ey, 2024-10-31:
Correct, though I was under the impression there was no need for the flash to be read during runtime. At least that's how I implemented my own software, but I suppose somewhere in rmk or embassy it's being read, so I'll have to look into that a bit

HaoboGu, 2024-10-31:
That depends you put your program. If you run your program in RAM, there should be no flash reading; But if you're running your program by XIP, then the flash is read while executing.

jackbrad1ey, 2024-11-01:
Do you know if there's any relevant settings in embassy or rmk related to this? I haven't been able to find much, I'll probably have to dig a little more to figure out where it's trying to read from the flash and if I can change it. Worst case I suppose I'll just have a dedicated boot key

HaoboGu, 2024-11-01:
I don't know.. it's defined in linker script maybe?