rmk-rs/rmk #122: One-shot key support

HaoboGu, 2024-11-02:
Reference:

- zmk sticky key: https://zmk.dev/docs/keymaps/behaviors/sticky-key
- qmk one-shot key: https://docs.qmk.fm/one_shot_keys

pcasotti, 2024-11-02:
I can try to implement this, have you already started on that?

I can think of two methods to implement this, one is processing the following events inside `process_key_action_oneshot` (just like `process_key_action_tap_hold`), the other is to add state to the `Keyboard` struct and handle it in `process_key_change`.

HaoboGu, 2024-11-03:
I have not started yet, it would be great if you can do this! Personally I prefer the first one, aka process_key_action_oneshot which is similar with the tap hold implementation

PR #123 (2024-11-11): One-shot key support
Closes #122
closed 2024-11-11 by HaoboGu