rmk-rs/rmk #124: Tap hold implementation issue

jackbrad1ey, 2024-11-04:
I've noticed there's a slight issue in the logic for the tap hold implementation. Specifically, relating to this section of code in the `process_key_action_tap_hold` function:
```rust
                embassy_futures::select::Either::Second(e) => {
                    if e.row == key_event.row && e.col == key_event.col {
                        // If it's same key event and releasing within 200ms, trigger tap
                        if !e.pressed {
                            let elapsed = self.timer[col][row].unwrap().elapsed().as_millis();
                            debug!("TAP action: {}, time elapsed: {}ms", tap_action, elapsed);
                            self.process_key_action_tap(tap_action, key_event).await;

                            // Clear timer
                            self.timer[col][row] = None;
                        }
                    } else {
                        // A different key comes within the threshold, trigger hold + that key

                        // Process hold action first
                        self.process_key_action_normal(hold_action, key_event).await;

                        // The actual processing is postponed because we cannot do recursion on async function without alloc
                        // After current key processing is done, we can process events in queue until the queue is empty
                        if self.unprocessed_events.push(e).is_err() {
                            warn!("unprocessed event queue is full, dropping event");
                        }
                    }
                }
```
Correct me if I'm wrong, but the `else` here can also be triggered by a key release event? What I've noticed is that if I'm typing fast, the tap of a tap hold key fails to trigger, this is because I press the tap hold key before releasing another key, so when I then release the other key it causes the code to go into this `else` condition, and then when I release the tap hold key (expecting the tap action to occur) it doesn't do anything since the else triggered the `hold` action.

I'm not quite familiar enough with Rust to make a PR with a fix, but in my head it sounds like what we'd want to do is loop until we timeout or get a _keypress_ event. If we get a key release event we should just process that action normally and then go back to waiting for a timeout or keypress.

HaoboGu, 2024-11-05:
Yes, you're right. There are some issues on current implementation of tap_hold. But processing keypress only might not be enough(see #118).

RMK implements key scanning and processing quite different from qmk/zmk. I want to find a better default config that is good for home row mods. If you have any idea, please share to me! 😄

HaoboGu, 2024-11-11:
@jackbrad1ey Could you try #125? This PR should address the tap/hold issue

jackbrad1ey, 2024-11-12:
Seems to be some issues, if I press a tap/hold key immediately after pressing another key (even if I quickly release that other key first) it triggers the tap action instantly and doesn't seem to wait to see if I will hold the key down. I have to wait for a short delay after pressing or releasing any other key before I can use the hold functionality.

HaoboGu, 2024-11-12:
> Seems to be some issues, if I press a tap/hold key immediately after pressing another key (even if I quickly release that other key first) it triggers the tap action instantly and doesn't seem to [...]

Yes, this is what [this config](https://github.com/urob/zmk-config?tab=readme-ov-file#timeless-homerow-mods) refers to. Maybe I should add an option to disable HRMs behavior?

jackbrad1ey, 2024-11-13:
Ah okay I see what it's intended for, I think ideally yeah an option to disable it would be nice. Personally I'll stick to the previous code, I use tap/hold on my enter key to double it as a layer shift and the old behaviour is a bit better. Though still a little odd with the issue that appears when typing fast, but it's not too noticeable.

HaoboGu, 2024-11-13:
> Ah okay I see what it's intended for, I think ideally yeah an option to disable it would be nice. Personally I'll stick to the previous code, I use tap/hold on my enter key to double it as a layer [...]

I'll add an option for it :P

HaoboGu, 2024-11-18:
> I'll add an option for it :P

Added.

PR #125 (2024-12-11): Refactor tap/hold processing, add better HRMs support out-of-the-box
Current tap/hold implementation has some issues, see #118, #124. This PR proposes an approach which addresses the issue, as well as HRM config.

The proposed solution is actually a mixture of [urob's zmk-config](https://github.com/urob/zmk-config), [Achordion](https://getreuer.info/posts/keyboards/achordion/index.html) and [sm_td](https://github.com/stasmarkin/sm_td). Many respect to the authors! 

The following is a simple description of new tap/hold processing:
When the keyboard processes tap/hold keys, it checks `prior_idl [... cut]
closed 2024-12-11 by HaoboGu