rmk-rs/rmk #130: Keymap doesn't update after flashing new firmware

mangkoran, 2024-11-09:
Thank you for your works on this firmware. I just found this last week and it's very straightforward compared to QMK.

I use RP2040 Pro Micro (16M flash). I found an issue where my keymap is not updated after I flash new firmware that was built after I have modified `keyboard.toml` with new keymap. Please let me know if you need something to help with this issue.

`keyboard.toml`
```toml
# Documentation: https://haobogu.github.io/rmk/keyboard_configuration.html
[keyboard]
name = "Morizon"
product_name = "Morizon"
vendor_id = 0xDEAD
product_id = 0xBEEF
manufacturer = "Steven Karrmann"
chip = "rp2040"

[matrix]
# TODO: Update input/output gpio name
# Input and output pins are mandatory
input_pins = [
  "PIN_0",
  "PIN_1",
  "PIN_2",
  "PIN_3",
  "PIN_4",
  "PIN_5",
  "PIN_6",
  "PIN_7",
  "PIN_8",
  "PIN_9",
]
output_pins = [
  "PIN_28",
  "PIN_27",
  "PIN_26",
  "PIN_22",
  "PIN_20",
  "PIN_23",
  "PIN_21",
]
# WARNING: Currently row2col/col2row is set in RMK's feature gate, configs here do nothing actually
# row2col = true

[layout]
rows = 10 # 4
cols = 7 # 3
layers = 1
keymap = [
  [
    [
      "Grave",
      "Kc1",
      "Kc2",
      "Kc3",
      "Kc4",
      "Kc5",
      "__",
    ],
    [
      "Escape",
      "Q",
      "W",
      "E",
      "R",
      "T",
      "__",
    ],
    [
      "Tab",
      "A",
      "S",
      "D",
      "F",
      "G",
      "__",
    ],
    [
      "LShift",
      "Z",
      "X",
      "C",
      "V",
      "B",
      "__",
    ],
    [
      "LCtrl",
      "__",
      "LGui",
      "LAlt",
      "__",
      "Space",
      "__",
    ],
    [
      "__",
      "Kc6",
      "Kc7",
      "Kc8",
      "Kc9",
      "Kc0",
      "Minus",
    ],
    [
      "__",
      "Y",
      "U",
      "I",
      "O",
      "P",
      "Backspace",
    ],
    [
      "__",
      "H",
      "J",
      "K",
      "L",
      "Semicolon",
      "Quote",
    ],
    [
      "__",
      "N",
      "M",
      "Comma",
      "Dot",
      "Slash",
      "Enter",
    ],
    [
      "__",
      "Space",
      "__",
      "Left",
      "Down",
      "UP",
      "Right",
    ],
  ],
]

[light]
# All light pins are high-active by default, uncomment if you want it to be low-active
# capslock.pin = "PB2"
# capslock.low_active = true
# scrolllock.pin = "PA3"
# scrolllock.low_active = true
# Just ignore if no light pin is used for it
# numslock.pin = "PA5"
# numslock.low_active = true

[storage]
# Storage feature is enabled by default
# enabled = false

# !!! Warning: RMK uses last 2 sectors of your microcontroller's flash by default. You have to ensure that those flash space is not used.
# If you're using a bootloader which puts itself at the end of the flast, like Adafruit_nRF52_Bootloader, RMK will break the bootloader.
# Setting the `start_addr` manually to avoid this. See: https://haobogu.github.io/rmk/storage.html
```

`vial.json`
```json
{
  "name": "Morizon",
  "vendorId": "0xDEAD",
  "productId": "0xBEEF",
  "lighting": "none",
  "matrix": {
    "rows": 10,
    "cols": 7
  },
  "layouts": {
    "keymap": [
      [
        "0,0",
        "0,1",
        "0,2",
        "0,3",
        "0,4",
        "0,5",
        {
          "x": 2
        },
        "5,1",
        "5,2",
        "5,3",
        "5,4",
        "5,5",
        "5,6"
      ],
      [
        "1,0",
        "1,1",
        "1,2",
        "1,3",
        "1,4",
        "1,5",
        {
          "x": 2
        },
        "6,1",
        "6,2",
        "6,3",
        "6,4",
        "6,5",
        "6,6"
      ],
      [
        "2,0",
        "2,1",
        "2,2",
        "2,3",
        "2,4",
        "2,5",
        "2,6",
        "7,0",
        "7,1",
        "7,2",
        "7,3",
        "7,4",
        "7,5",
        "7,6"
      ],
      [
        "3,0",
        "3,1",
        "3,2",
        "3,3",
        "3,4",
        "3,5",
        "3,6",
        "8,0",
        "8,1",
        "8,2",
        "8,3",
        "8,4",
        "8,5",
        "8,6"
      ],
      [
        "4,0",
        "4,1",
        "4,2",
        "4,3",
        "4,4",
        "4,5",
        "4,6",
        "9,0",
        "9,1",
        "9,2",
        "9,3",
        "9,4",
        "9,5",
        "9,6"
      ]
    ]
  }
}
```

`memory.x`
```
MEMORY {
    BOOT2 : ORIGIN = 0x10000000, LENGTH = 0x100
    FLASH : ORIGIN = 0x10000100, LENGTH = 16384K - 0x100
    RAM   : ORIGIN = 0x20000000, LENGTH = 264K
}

EXTERN(BOOT2_FIRMWARE)

SECTIONS {
    /* ### Boot loader */
    .boot2 ORIGIN(BOOT2) :
    {
        KEEP(*(.boot2));
    } > BOOT2
} INSERT BEFORE .text;
```

`main.rs`
```rust
#![no_std]
#![no_main]

use rmk::macros::rmk_keyboard;

#[rmk_keyboard]
mod keyboard {}
```

Related https://github.com/HaoboGu/rmk/issues/41

HaoboGu, 2024-11-09:
yeah this is intended because I suppose that people rarely change keymap via reflashing the firmware  and don't want their edited keymap overwritten when reflashing/updating the firmware.

mangkoran, 2024-11-09:
Fair. AFAIK Vial still doesn't have keymap export yet so this might be useful for some. How about we provide CLI option when flashing so that user could choose to overwrite the keymap too?

HaoboGu, 2024-11-09:
Yes, I agree that having this can be great. 

But doing it is quite hard, because for different MCUs, the flashing CLI is different. I think you can implement this for your microcontroller first, and wait for other people to do for theirs when they need :D