rmk-rs/rmk #131: Vial Matrix Tester seems cannot be used

mangkoran, 2024-11-09:
It seems Vial Matrix Tester is cannot be used. It doesn't respond to any input.

[image]

HaoboGu, 2024-11-09:
good find, it's just not implemented yet :(

whitelynx, 2024-11-11:
@HaoboGu any idea what pieces of the protocol need to be implemented in order for this to work? I'm not that familiar with Vial's protocol, and so far I haven't found the specific command(s) related to this feature.

HaoboGu, 2024-11-11:
> @HaoboGu any idea what pieces of the protocol need to be implemented in order for this to work? I'm not that familiar with Vial's protocol, and so far I haven't found the specific command(s) related to this feature.

It's `SwitchMatrixState` event, the qmk implement [this way](https://github.com/vial-kb/vial-qmk/blob/4b2ae32e871c07e50703f0af591c479e66dc7eab/quantum/via.c#L249). However, it's hard to do it in Rust. I don't know if there's a easy way to implement this 🥲

tib888, 2025-03-05:
Is there a plan to support "NKRO"?
Because that (if I know right) NKRO also needs the state storage of every keys, not only 6 + modifiers.
But of course there is a difference: vial needs the phisical keys, while NKRO needs the "after mapped to FullSize" keys pressed state... 

Also, I was wondering why register_keycode() has this strange code inside:
```
        // First, find the key event slot according to the position
        let slot = self.registered_keys.iter().enumerate().find_map(|(i, k)| {
            if let Some((row, col)) = k {
                if key_event.row == *row && key_event.col == *col {
                    return Some(i);
                }
            }
            return None;
        });
```
Why is that dealing with row/col positions???
Maybe to use it for VIAL matrix tester?

pcasotti, 2025-03-25:
Adding a note here. Before we add support for vial matrix tester we need to implement the vial security feature, otherwise a malicious program can use the vial protocol to implement a keylogger

HaoboGu, 2025-08-21:
Closed by #516

closed 2025-08-21 by HaoboGu