rmk-rs/rmk #133: Documentation links are partly broken

thomasaarholt, 2024-11-10:
Best experienced by first going to one of the docs pages:
https://haobogu.github.io/rmk/keyboard_configuration.html

Then click on one of the menu options to the right, like "Overview". After the page loads, try clicking on any of the menu options again (including "Overview"). The link is now broken.

The correct link is:
`https://haobogu.github.io/rmk/user_guide/1_guide_overview.html`
The broken link is:
`https://haobogu.github.io/rmk/user_guide/user_guide/1_guide_overview.html`

Note the double `user_guide/`.

HaoboGu, 2024-11-10:
I can confirm it's a bug of documentation site. I'll fix it soon

decaday, 2024-11-12:
This issue is likely caused by the update to index.hbs in mdBook v0.4.41.
https://github.com/rust-lang/mdBook/blob/90960126e8b2da1dc1a7e460c2bdeb8073f1b81e/CHANGELOG.md?plain=1#L11-L14

I have attempted to fix it in #141.

closed 2024-11-12 by HaoboGu