rmk-rs/rmk #134: Why there is no Mod-Tap

lqlklu, 2024-11-10:
zmk has a Behavior named Mod-Tap([https://zmk.dev/docs/keymaps/behaviors/mod-tap](https://zmk.dev/docs/keymaps/behaviors/mod-tap)), but it seems no such option in rmk config file or rust macros.
I tried this code in rust, I think it should work, but it not.
```rust
/// in keymap.rs

macro_rules! mt {
    ($a: ident, $b: ident) => {
        KeyAction::TapHold(Action::Key(KeyCode::$a), Action::Key(KeyCode::$b))
    };
}

/// and I use it in `fn get_default_keymap()`
mt!(LCtrl, Escape)
```
It successfully compiled, but not work after flashing.

HaoboGu, 2024-11-10:
Yeah, macro for mod-tap is missing. Could you try this?

```rust
mt!(Escape, LCtrl)  // The first argument is the "tap" action, the second is hold
```

lqlklu, 2024-11-10:
It still not work.
There is no new device appear on Windows Device Manager after keyboard plugin, and all key on keyboard not work.
I don't known what happened because I have no debug probe.
Full code here: [https://github.com/lqlklu/ergox-keyboard/tree/main/firmware/ergox_rs](https://github.com/lqlklu/ergox-keyboard/tree/main/firmware/ergox_rs)

HaoboGu, 2024-11-10:
let me have a try

HaoboGu, 2024-11-10:
Oh sorry I forgot it, there's mod-tap in RMK, named `ModifierTapHold`: https://docs.rs/rmk/latest/rmk/action/enum.KeyAction.html#variant.ModifierTapHold

Normal tap/hold is not supported right now.

lqlklu, 2024-11-11:
> Oh sorry I forgot it, there's mod-tap in RMK, named `ModifierTapHold`: https://docs.rs/rmk/latest/rmk/action/enum.KeyAction.html#variant.ModifierTapHold

`ModifierTapHold` still successfully compiled but not work.
I tried this code:
```rust
KeyAction::ModifierTapHold(Action::Key(KeyCode::Escape), ModifierCombination::new_from(false,false,false,false,true))
```
Full code here updated: https://github.com/lqlklu/ergox-keyboard/tree/main/firmware/ergox_rs

HaoboGu, 2024-11-11:
@lqlklu Could you try to edit the keymap using https://vial.rocks ?

Editing the keymap in the source file won't affect the keymap, see: #130

lqlklu, 2024-11-11:
It shows “there is no compatible device found“(找不到兼容设备。). Actually, there is no such device before and after plugin in Windows Device Manager, so it is impossible show up in vial.

lqlklu, 2024-11-11:
> It shows “there is no compatible device found“(找不到兼容设备。). Actually, there is no such device before and after plugin in Windows Device Manager, so it is impossible show up in vial.

By the way, when using `keyboard.toml` or normal rust code rather than rust code using `ModifierTapHold`, It works well with vial.

HaoboGu, 2024-11-12:
🤔 Could you try to erase your flash, increase the `num_sectors` value and then re-flash the firmware?

If there's no keyboard device shown on your PC, the firmware dead before keyboard starting, so I (randomly) guess it's a storage error. (I don't have an rp2040 at my hand, cannot test it by myself 🥲

HaoboGu, 2024-11-12:
> but I don't known how to increase `num_sectors`

see docs here: https://haobogu.github.io/rmk/keyboard_configuration.html#storage

lqlklu, 2024-11-12:
It works! Thank you so much.
I wrote this code:
```rust
    let keyboard_config = RmkConfig {
        usb_config: keyboard_usb_config,
        vial_config,
        storage_config: StorageConfig {
            start_addr: 0x00000000,
            num_sectors: 4,
        },
        ..Default::default()
    };
```
and it finally works with both keyboard tester and vial.

closed 2024-11-12 by lqlklu

HaoboGu, 2024-11-12:
And, for rp2040, you can increase it to larger number, like 16 or 32, ensuring there's enough space for storage