rmk-rs/rmk #149: Customizable keycode behavior

MixedMatched, 2024-11-15:
It might be helpful to add a custom handler for key input so that a user can have custom actions that can't be done by existing keycodes. QMK does this with [process_record_user](https://docs.qmk.fm/custom_quantum_functions).

This could probably be implemented by having the user supply an optional function pointer with the signature `KeyAction -> bool` that would be run before the match statement in `Keyboard::process_key_change`. The returned boolean would then control whether the normal action handler is run after the custom handler. This would maybe also require exposing other fields or channels so that the function could add keycodes to be processed or trigger other parts of the firmware.

HaoboGu, 2024-11-16:
I agree, we should have this

tib888, 2025-03-05:
How is this related to the ZMK "behavior-mod-morph"?

//this is used to enter "á" or "Á" depending on the shift's state:
hun_aaAA_: hun_aaAA_ {
            compatible = "zmk,behavior-mod-morph";
            #binding-cells = <0>;
            bindings = <&hun_aa>, <&hun_AA_>;
            mods = <(MOD_LSFT|MOD_RSFT)>;
        };

Or Kanata "fork" (this config is MUCH nicer BTW!)?

áÁ (fork (unicode á) (unicode Á) (lsft rsft))

If this issue is about something else then I'll add a separate issue to support such a state-fork behavior simirarly to Combos.
(best would be to be able to fork depending not only based on modifier states, but also based on XXXlock active states and maybe layer activation states).

I guess VIAL has no way to configure it, but I still need it.