rmk-rs/rmk #156: Add an option making it possible to update keymap when re-flashing

HaoboGu, 2024-11-30:


closed 2024-12-02 by HaoboGu

mangkoran, 2025-01-06:
Is this ticket similar to https://github.com/HaoboGu/rmk/issues/130 ?

HaoboGu, 2025-01-06:
> Is this ticket similar to #130 ?

Yes, but when enabling `clear_storage` option, the storage would be erased on every reboot. It can only be used for testing situation

HaoboGu, 2025-01-06:
> If so, could we have another config where we could clear the storage during firmware flash only? So we could have two different config for two different purpose.

I think it's impossible. The firmware doesn't know this.

HaoboGu, 2025-01-06:
zmk does it by having a separate firmware `reset-config`. If you want to clear the storage, you have to flash the `reset-config` firmware first, then flash the keyboard firmware

mangkoran, 2025-01-06:
> I think it's impossible. The firmware has no way to know whether current boot is right after a re-flashing

You are correct. At first I thought it's possible to implement in flasher/CLI, but then I forgot that I use UF2 bootloader to flash the firmware. Thank you for answering my question. I've been using RMK ever since I found it <3

PR #161 (2024-12-02): feat(storage): add `clear_storage` option
Closes #156