rmk-rs/rmk #157: No bluetooth device after flashing seed studio nrf52840 for split BLE

thomasaarholt, 2024-11-30:
Hi!
Thank you so much for creating this, I'm really enjoying following these developments!

**Short version:** When I connect the two keyboard halves to usb power (either on my mac or via a usb charger), I do not see the keyboard in the bluetooth menu on my Mac.

**Background**
I made an RMK dummy 1-key bluetooth (non-split) keyboard last week with only the letter "A", and that worked perfectly (I could see it in the bluetooth menu), so I decided to build a split keyboard. I have now finished the first half keyboard. However, I'm running into some issues when flashing it.

For this issue, I recreated a template from `cargo generate --git https://github.com/HaoboGu/rmk-template` to make sure I was running the latest version. I copied in my `vial.json` and entered the correct info in `keyboard.toml`, and then ran `cargo make uf2 --release`. The nrf52840 is a seeed studio nrf52840, and it is running Softdevices S140 7.3.0.

I have pushed [all code and files here](https://github.com/thomasaarholt/rmk-nrf52840-split-ble).

**The issue**
I do not have a debug probe, so I set the seeed studio in bootloader mode (pressed reset button twice), and then copied over the `rmk-nrf52840-split-ble-central.uf2` (this give me a copy error, which I have learnt is just because the microcontroller restarts as soon as it gets the file, before it has told the computer that the file copied successfully). I then did the same for the peripheral uf2, on the other microcontroller.

Now, when I connect the two to usb power (either on my mac or via a usb charger), I do not see any RMK Keyboard in the bluetooth menu. Do you have any advice?

HaoboGu, 2024-12-01:
can you see any USB device? By default if USB is connected, the BLE would not advertise

thomasaarholt, 2024-12-01:
I'm connecting it to an Apple USB charger instead of my mac. BLE is not advertising. Should I expect BLE to advertise? Or does it have to be connected to power via the power pins on the microcontroller? 

If the USB charger is what is expected, could the microcontroller be thinking that the USB charger is a computer, and thus not advertise BLE? 

With my one-key keyboard, I observed that BLE did advertise last week, but it happened a bit suddenly, on USB power. A bit like I maybe wasn't waiting long enough for the device to boot, or something.

If I only flash one half of the keyboard, using the central uf2 and connect that to USB charger power - without connecting the peripheral - should I expect BLE to advertise? Or does it need to talk to the second microcontroller in order to advertise?

HaoboGu, 2024-12-01:
yeah the default setting is USB-priority, which means if a USB cable is connected, BLE would not advertise. You can change it using 'switch output': https://haobogu.github.io/rmk/wireless.html#multiple-profile-support

thomasaarholt, 2024-12-01:
This makes total sense! Just so I’m sure - I can’t change the default, but you are suggesting that I map a key to «switch output» so that I can press that button to stay in BLE mode?

HaoboGu, 2024-12-01:
> This makes total sense! Just so I’m sure - I can’t change the default, but you are suggesting that I map a key to «switch output» so that I can press that button to stay in BLE mode?

yes. you can set it in USB mode and using [vial](vial.rocks). The key is under "user" tab.