rmk-rs/rmk #158: Setting col2row and row2col, and what is default?

thomasaarholt, 2024-11-30:
[I took this commented out line](https://github.com/thomasaarholt/rmk-nrf52840-split-ble/blob/49df4016e5151bf666d592c150407f29c426244b/keyboard.toml#L15) as implying that the diode should allow current from the row to the column. Is that a correct interpretation? If not, how can I change this in [cargo.toml](https://github.com/thomasaarholt/rmk-nrf52840-split-ble/blob/49df4016e5151bf666d592c150407f29c426244b/Cargo.toml#L13)?

HaoboGu, 2024-12-01:
@thomasaarholt col2row is the default, if you want to use row2col matrix, you can add the feature as the following:

```toml
rmk = { version = "0.4", default-features = false, features = [
    "nrf52840_ble",
    "row2col",
    "rapid_debouncer",
    "async_matrix",
] }
```

thomasaarholt, 2024-12-01:
Aha! Thank you!

I couldn't find that feature amongst the features on the [rust docs features page](https://docs.rs/crate/rmk/0.4.4/features), so I wasn't sure. 

Oh, but when I try to compile with those settings, I get an error:

```rust
❯ cargo build --release
error: failed to select a version for `rmk`.
    ... required by package `rmk-nrf52840-split-ble v0.0.1 (/Users/thomas/codes/rmk/rmk-nrf52840-split-ble)`
versions that meet the requirements `^0.4` (locked to 0.4.4) are: 0.4.4

the package `rmk-nrf52840-split-ble` depends on `rmk`, with features: `row2col` but `rmk` does not have these features.

failed to select a version for `rmk` which could resolve this conflict
```

thomasaarholt, 2024-12-01:
Ah, could it be that there is no `row2col` feature, but that the _lack_ of the _default_ `col2row` feature implies that `row2col` is active?

So adjusting your code suggestion:
```toml
rmk = { version = "0.4", default-features = false, features = [
    "nrf52840_ble",
     # "row2col", # remove this
    "rapid_debouncer",
    "async_matrix",
] }

HaoboGu, 2024-12-01:
ohhhh sorry! I forgot that has been removed, you're right, use 'default-features = false' is enough

HaoboGu, 2024-12-02:
I'm closing this, if you still have problems, feel free to reopen it :)

closed 2024-12-02 by HaoboGu