rmk-rs/rmk #159: BLE flash not working with after cargo update

thomasaarholt, 2024-12-01:
Hello!

I observe that when making a new rmk config using rmk 0.4.4, BLE does not advertise and I cannot find the keyboard in the Bluetooth menu on my mac. When I use a keyboard config from last week with rmk 0.4.1, it does work.

## Here is my repro:
**Working**
I have a working rmk keyboard config in [this repo](https://github.com/thomasaarholt/rmk_oneletter).

I run:

```bash
cargo clean
cargo make uf2 --release
cp rmk.uf2 /Volumes/XIAO-SENSE/. # microcontroller restarts
```
While the keyboard is still connected to my mac, "RMK Keyboard" appears in the bluetooth search menu.

**Not working**
Same repo, but instead I run cargo update first.
```bash
cargo clean
cargo update
cargo make uf2 --release
cp rmk.uf2 /Volumes/XIAO-SENSE/. # microcontroller restarts
```

Keyboard does not advertise BLE. This is the same behaviour that I observe when I make a new keyboard using the cargo generator.

HaoboGu, 2024-12-01:
@thomasaarholt As I mentioned in https://github.com/HaoboGu/rmk/issues/157#issuecomment-2509721126, changing default output mode might solve your problem. This was introduced in v0.4.2, you used v0.4.1, so the BLE advertising was "fixed" in latest versions LOL

thomasaarholt, 2024-12-02:
Yup, this will be it! Thanks!

closed 2024-12-02 by thomasaarholt