rmk-rs/rmk #160: direct_pin split

willpuckett, 2024-12-01:
When attempting to build for a direct_pin split, I'm receiving the following error:

```
   Compiling embassy-nrf v0.2.0
   Compiling nrf-softdevice v0.1.0 (https://github.com/embassy-rs/nrf-softdevice?rev=d5f023b#d5f023ba)
   Compiling rmk v0.4.4
error: Parse `keyboard.toml` error: missing field `input_pins`
 --> src/peripheral.rs:6:1
  |
6 | #[rmk_peripheral(id = 0)]
  | ^^^^^^^^^^^^^^^^^^^^^^^^^
  |
  = note: this error originates in the attribute macro `rmk_peripheral` (in Nightly builds, run with -Z macro-backtrace for more info)

error: Parse `keyboard.toml` error: missing field `input_pins`
 --> src/central.rs:6:1
  |
6 | #[rmk_central]
  | ^^^^^^^^^^^^^^
  |
  = note: this error originates in the attribute macro `rmk_central` (in Nightly builds, run with -Z macro-backtrace for more info)

error: `#[panic_handler]` function required, but not found

error: could not compile `ehrbl` (bin "peripheral") due to 2 previous errors
warning: build failed, waiting for other jobs to finish...
error: could not compile `ehrbl` (bin "central") due to 2 previous errors
```

I built one half of the board successfully last night as a numpad.

My `keyboard.toml` for the split:

```
# Documentation: https://haobogu.github.io/rmk/keyboard_configuration.html
[keyboard]
name = "ehrbl"
product_name = "ehrbl"
vendor_id = 0x4c4b
product_id = 0x4643
manufacturer = "haobo"
chip = "nrf52840"
board = "XIAO BLE"

[layout]
# WARNING: Currently row2col/col2row is set in RMK's feature gate, configs here do nothing actually
# row2col = true
rows = 4
cols = 8
layers = 3
keymap = [
    [
        ["B", "F", "D", "L", "P", "O", "U", "Y"],
        ["C", "S", "T", "R", "N", "A", "I", "H"],
        ["V", "W", "G", "M", "K", ",", ".", "'"],
        ["_", "_", "LT(2, Backspace)", "LT(1, Space)", "E", "Enter", "_", "_"]
    ],
    [
        ["KpEnter", "Kp7", "Kp8", "Kp9", "KbVolumeDown", "KbMute", "KbVolumeUp", "_"],
        ["Kp0", "Kp4", "Kp5", "Kp6", "Left", "Down", "UP", "Right"],
        ["KpDot", "Kp1", "Kp2", "Kp3", "Home", "PageDown", "PageUp", "End"],
        ["_", "_", "_", "_", "_", "K_CANCEL", "_", "_"]
    ],
    [
        ["F12", "F7", "F8", "F9", "_", "_", "_", "_"],
        ["F11", "F4", "F5", "F6", "_", "_", "_", "_"],
        ["F10", "F1", "F2", "F3", "_", "_", "_", "_"],
        ["_", "_", "_", "_", "_", "_", "_", "_"]
    ],
]

[light]
# All light pins are high-active by default, uncomment if you want it to be low-active
# capslock.pin = "PB2"
# capslock.low_active = true
# scrolllock.pin = "PA3"
# scrolllock.low_active = true
# Just ignore if no light pin is used for it
# numslock.pin = "PA5"
# numslock.low_active = true

[storage]
# Storage feature is enabled by default
# enabled = false

# !!! Warning: RMK uses last 2 sectors of your microcontroller's flash by default. You have to ensure that those flash space is not used.
# If you're using a bootloader which puts itself at the end of the flast, like Adafruit_nRF52_Bootloader, RMK will break the bootloader.
# Setting the `start_addr` manually to avoid this. See: https://haobogu.github.io/rmk/storage.html

[ble]
enabled = true

[split]
connection = "ble"

[split.central]
rows = 4
cols = 4
row_offset = 0
col_offset = 0
ble_addr = [0x18, 0xe2, 0x21, 0x80, 0xc0, 0xc7]
matrix_type = "direct_pin"
direct_pins = [
    ["P0_28", "P0_03", "P0_02", "P1_10"], 
    ["P0_04", "P1_14", "P0_29", "P1_15"],
    ["P1_13", "P0_05", "P1_11", "P1_12"],
    ["__", "__", "P0_09", "P0_10"]
]
# `direct_pin_low_active` is optional. Default is `true`.
direct_pin_low_active = true

[[split.peripheral]]
rows = 4
cols = 4
row_offset = 0
col_offset = 4
ble_addr = [0x7e, 0xfe, 0x73, 0x9e, 0x66, 0xe3]
matrix_type = "direct_pin"
direct_pins = [
    ["P1_10", "P1_15", "P0_29", "P1_14"], 
    ["P0_04", "P1_13", "P0_05", "P1_12"],
    ["P0_28", "P1_11", "P0_09", "P0_10"],
    ["P0_02", "P0_03", "__", "__"]
]
# `direct_pin_low_active` is optional. Default is `true`.
direct_pin_low_active = true

# [behavior]
# # Enable HRM 
# tap_hold = { enable_hrm = true, prior_idle_time = "120ms", hold_timeout = "250ms", post_wait_time = "50ms"}
# # Disable HRM, you can safely ignore any fields if you don't want to change them
# # tap_hold = { enable_hrm = false, hold_timeout = "200ms" }
```

I first tried building using the hrm branch, then when it failed, changed back to rmk 0.4.1 and ran `cargo clean` before attempting to compile again, but received the same error....

willpuckett, 2024-12-01:
oops didn't see #140... I'll try it now...

willpuckett, 2024-12-01:
After fixing a few typos, it's successfully building now with PR #140...

It seems that using more than 1 "_" in a single line of a direct_pins array errors, both with rmk 0.4.1 and in PR #140... Will try flashing shortly.

HaoboGu, 2024-12-03:
I'm closing this since #163 has been merged. If you have other issues, feel free to open a new issue :)

closed 2024-12-03 by HaoboGu