rmk-rs/rmk #164: Confusion about split row/col offset

thomasaarholt, 2024-12-03:
Hi again!
I'm a bit confused by the docs for the split keyboard, including the examples. 

For a split keyboard that has two identical 5x5 keys, for a total of 5 rows and 10 columns together, what is the correct col and row offset for the peripheral?

I would imagine it should be as per the following (most relevant bit is at the bottom of the code field, scroll):
```
[layout]
rows = 5
cols = 10
layers = 1
keymap = [
    [
        ["Q", "W", "E", "R", "T", "Y", "U", "I", "O", "P"],
        ["A", "S", "D", "F", "G", "H", "J", "K", "L", "Quote"],
        ["Z", "X", "C", "V", "B", "N", "M", "Comma", "Dot", "Slash"],
        ["Kc1", "Kc2", "Kc3", "Kc4", "Kc5", "Kc6", "Kc7", "Kc8", "Kc9", "Kc0"],
        ["A", "B", "C", "D", "E", "F", "G", "H", "I", "J"],
    ],
]

[split.central]
rows = 5
cols = 5
row_offset = 0
col_offset = 0

[[split.peripheral]]
rows = 5
cols = 5
row_offset = 0 # <--- this bit is most interesting
col_offset = 5 # <--- this bit is most interesting
```

But in the [nrf ble split example](https://github.com/HaoboGu/rmk/blob/main/examples/use_config/nrf52840_ble_split/keyboard.toml), we have the following, where row offset is set to 2 for the peripheral.

```toml
[layout]
rows = 4
cols = 3

[split.central]
rows = 2
cols = 2
row_offset = 0
col_offset = 0

[[split.peripheral]]
rows = 2
cols = 1
row_offset = 2 # Is this incorrect? The row offset is zero, isn't it?
col_offset = 2
```

I have also looked at the [real life example](https://github.com/whitelynx/dactyl-lynx-rmk/blob/main/keyboard.toml) linked in the github repo readme, and that seems to do what I am doing, but also does strange things like having 14 rows and 6 cols, so I tought it best to ask.

HaoboGu, 2024-12-03:
Your setting is correct. 

The row offset in the example is set to 2 because there are 4 rows in total. In the example, not all keys in the matrix are covered, that might be confusing.

HaoboGu, 2024-12-11:
@thomasaarholt Did you solve this?

thomasaarholt, 2024-12-12:
Yes! Happy to close this. I do recommend adding a section in the user guide about row2col / col2row and how to configure it, though.

closed 2024-12-12 by HaoboGu

HaoboGu, 2024-12-13:
I added some documentation