rmk-rs/rmk #167: ext power control & ble led

jsenjobs, 2024-12-03:
Can I set a control pin for ext power control and  set a LED pin for Bluetooth wake-up like zmk.

HaoboGu, 2024-12-04:
> Can I set a control pin for ext power control and set a LED pin for Bluetooth wake-up like zmk.

Could you explain a little bit more about your usage?

jsenjobs, 2024-12-05:
> > Can I set a control pin for ext power control and set a LED pin for Bluetooth wake-up like zmk.
> 
> Could you explain a little bit more about your usage?

### ext power control:
like nicenano can control rgb & display etc on/off.
[image]

### ble led：
like nicenano when wakeup rmk/zmk , then led can blink。
[image]

HaoboGu, 2025-08-21:
Track this in #261

closed 2025-08-21 by HaoboGu