rmk-rs/rmk #171: the debouncer of split direct pins

drindr, 2024-12-08:
the debouncer for the split direct pin matrix seems wrong.
when i'm trying to build a non-square matrix, the debouncer panic

https://github.com/HaoboGu/rmk/blob/a212bbba6d328d6fa311bd77a9b90768ec7764b9/rmk/src/split/central.rs#L74-L82

> INFO  debouncing idx: 0 0
└─ rmk::split::central::{impl#3}::scan::{async_fn#0} @ /home/drin/workspace/practoy/rmk/rmk/src/split/central.rs:613
INFO  debouncing idx: 1 0
└─ rmk::split::central::{impl#3}::scan::{async_fn#0} @ /home/drin/workspace/practoy/rmk/rmk/src/split/central.rs:613
INFO  debouncing idx: 2 0
└─ rmk::split::central::{impl#3}::scan::{async_fn#0} @ /home/drin/workspace/practoy/rmk/rmk/src/split/central.rs:613
INFO  debouncing idx: 3 0
└─ rmk::split::central::{impl#3}::scan::{async_fn#0} @ /home/drin/workspace/practoy/rmk/rmk/src/split/central.rs:613
INFO  debouncing idx: 4 0
└─ rmk::split::central::{impl#3}::scan::{async_fn#0} @ /home/drin/workspace/practoy/rmk/rmk/src/split/central.rs:613
ERROR panicked at /home/drin/workspace/practoy/rmk/rmk/src/debounce/default_bouncer.rs:61:54:
index out of bounds: the len is 4 but the index is 4

everything is ok if i swap the col and row in the definition of debouncer.

and i notice that in the `direct_pin.rs`, there is only the definition about col2row.

https://github.com/HaoboGu/rmk/blob/a212bbba6d328d6fa311bd77a9b90768ec7764b9/rmk/src/direct_pin.rs#L169-L172

i'm curious about which one is correct.
the row and col in different places are a bit confusing.
the in and out pins of direct pin matrix are a bit confusing as well. 

@DivineGod

drindr, 2024-12-09:
> I checked the split-direct pin example, worked well with non-square matrix. Could you post your `keyboard.toml` here?

thanks for your quick reply.
my `keyboard.toml` is https://0x0.st/XhMw.toml

HaoboGu, 2024-12-09:
> should i add the feature col2row in dependencies item in Cargo.toml explicitly

If you don't have `default-features = false`, you don't need to add it explicitly.

HaoboGu, 2024-12-09:
I can repro it. I'll fix it soon

HaoboGu, 2024-12-09:
@drindr Could you try https://github.com/HaoboGu/rmk/pull/172 this branch?

drindr, 2024-12-09:
> @drindr Could you try #172 this branch?

everything looks good now. thanks.

PR #172 (2024-12-09): Fix direct pin matrix size error
Closes #171 

Direct pin doesn't need to be affected by `col2row` feature. This PR removes it as well.
closed 2024-12-09 by HaoboGu