rmk-rs/rmk #180: Unexpected Behavior on Direct Split

willpuckett, 2024-12-12:
I've been testing RMK on a direct split, and I'm having a few issues, hopefully due to user error...

I'm building from the main branch. For compatibility with direct split and improved hrms, my Cargo.toml has:

```
[dependencies]
rmk = { git = "https://github.com/HaoboGu/rmk",  features = [
...
```

I have also ensured the Xiaos are on the latest soft device.

After flashing RMK, the left (central) portion of the keyboard works as anticipated. The right half, however, only outputs keypresses on the top two of four rows. 

If I hold a key on the top two rows of the right (peripheral) side and press a previously non-working key on one of the two bottom rows, both keypresses are sent. Holding a key on the left side and pressing one of the misbehaving keys only outputs the left key.

I was afraid I might have misconfigured my row/col offsets, but referencing #164 it seems they are correct.

It also doesn't seem to be a hardware problem: the keys are working normally in my zmk build both before and after flashing rmk.

I double checked my split.peripheral.matrix and everything seems to be entered correctly.

Changing the key assignments of the misbehaving keys in Vial has no effect.

Any thoughts or directions are greatly appreciated! 

My current keyboard.toml:

```
# Documentation: https://haobogu.github.io/rmk/keyboard_configuration.html
[keyboard]
name = "ehrbl"
product_name = "ehrbl"
vendor_id = 0x4c4b
product_id = 0x4643
manufacturer = "haobo"
chip = "nrf52840"
board = "XIAO BLE"

[layout]
# WARNING: Currently row2col/col2row is set in RMK's feature gate, configs here do nothing actually
# row2col = true
rows = 4
cols = 8
layers = 3
keymap = [
    [
        ["B", "F", "D", "L", "P", "O", "U", "Y"],
        ["C", "S", "T", "R", "N", "A", "I", "H"],
        ["V", "W", "G", "M", "K", "Comma", "Dot", "Quote"],
        ["__", "__", "LT(2, Backspace)", "LT(1, Space)", "E", "Enter", "__", "__"]
    ],
    [
        ["KpEnter", "Kp7", "Kp8", "Kp9", "KbVolumeDown", "KbMute", "KbVolumeUp", "__"],
        ["Kp0", "Kp4", "Kp5", "Kp6", "Left", "Down", "UP", "Right"],
        ["KpDot", "Kp1", "Kp2", "Kp3", "Home", "PageDown", "PageUp", "End"],
        ["__", "__", "__", "__", "__", "__", "__", "__"]
    ],
    [
        ["F12", "F7", "F8", "F9", "__", "__", "__", "__"],
        ["F11", "F4", "F5", "F6", "__", "__", "__", "__"],
        ["F10", "F1", "F2", "F3", "__", "__", "__", "__"],
        ["__", "__", "__", "__", "__", "__", "__", "__"]
    ],
]

[light]

[storage]

[ble]
enabled = true

[split]
connection = "ble"

[split.central]
rows = 4
cols = 4
row_offset = 0
col_offset = 0
ble_addr = [0x18, 0xe2, 0x21, 0x80, 0xc0, 0xc7]

[split.central.matrix]
matrix_type = "direct_pin"
direct_pins = [
    ["P0_28", "P0_03", "P0_02", "P1_10"], 
    ["P0_04", "P1_14", "P0_29", "P1_15"],
    ["P1_13", "P0_05", "P1_11", "P1_12"],
    ["_", "_", "P0_09", "P0_10"]
]
# `direct_pin_low_active` is optional. Default is `true`.
direct_pin_low_active = true

[[split.peripheral]]
rows = 4
cols = 4
row_offset = 0
col_offset = 4
ble_addr = [0x7e, 0xfe, 0x73, 0x9e, 0x66, 0xe3]

[split.peripheral.matrix]
matrix_type = "direct_pin"
direct_pins = [
    ["P1_10", "P1_15", "P0_29", "P1_14"], 
    ["P0_04", "P1_13", "P0_05", "P1_12"],
    ["P0_28", "P1_11", "P0_09", "P0_10"],
    ["P0_02", "P0_03", "_", "_"]
]
# `direct_pin_low_active` is optional. Default is `true`.
direct_pin_low_active = true

[behavior]
# Enable HRM 
tap_hold = { enable_hrm = true, prior_idle_time = "150ms", hold_timeout = "280ms", post_wait_time = "25ms"}
# # Disable HRM, you can safely ignore any fields if you don't want to change them
# # tap_hold = { enable_hrm = false, hold_timeout = "200ms" }
```

HaoboGu, 2024-12-12:
The config file looks correct. Could you try to switch the peripheral rows: 

```
[split.peripheral.matrix]
matrix_type = "direct_pin"
direct_pins = [
    ["P0_28", "P1_11", "P0_09", "P0_10"],
    ["P0_02", "P0_03", "_", "_"],
    ["P1_10", "P1_15", "P0_29", "P1_14"], 
    ["P0_04", "P1_13", "P0_05", "P1_12"]
]
# `direct_pin_low_active` is optional. Default is `true`.
direct_pin_low_active = true
```

willpuckett, 2024-12-13:
Just tried making the above change. I ran `cargo clean`, erased the previous uf2 files, and then ran the standard build commands to produce new uf2s.

Flashing the new firmware, the top two peripherial rows now don't work, with the the exception of RT1 and RT2, which work normally. The top two rows do output characters when pressed in conjunction with a key from the bottom two peripherial rows.

The top two rows output keys assigned to the bottom two rows both in the keymap and Vial, and the bottom two rows output keys assigned to the top two rows, ie, the new peripherial keys output the keys assigned to the originally assigned direct pins.

I'd be happy to try and further debug this however possible. The controllers (xiao bles) are soldered onto the board, otherwise I would try erasing the chips with probe-rs. If that seems like the next reasonable step, let me know and I'll proceed in that direction...

HaoboGu, 2024-12-13:
You don't have to do it right now if it's complex. Instead, you can add 

```toml
[storage]
# Set `clear_storage` to true to clear all the stored info when the keyboard boots
clear_storage = true
```

to clear all stored info in the keyboard, including bond info, keymap, etc.

Could you try this and see whether the behavior changes? thanks!

willpuckett, 2024-12-14:
Just tried, no change… Also flashed opposite sides with central and peripheral (with clear_storage = true) and had same behavior.

HaoboGu, 2024-12-14:
This is super weird. 

Could you try again with removing `async_matrix` feature in Cargo.toml? I've ran out of my idea...

HaoboGu, 2024-12-17:
@willpuckett would you like to try #200？

willpuckett, 2024-12-18:
Hey absolutely I've been away from my desk but I'll give it a try later
tonight as soon as I get home.

Cheers!
WP

willpuckett, 2024-12-18:
closing resolved by #200

closed 2024-12-18 by willpuckett