rmk-rs/rmk #182: Vial not working with BLE

thomasaarholt, 2024-12-12:
I've just got my BLE keyboard working with rmk 0.4.4 (hooray!), but when I try to use vial to adjust the keymap, I get this error:

[image]

It still works with USB, so it isn't a problem for me, but it might be nice to add to the docs if this is intentional / only USB is supported for vial realtime updates.

HaoboGu, 2024-12-12:
It's a known issue. vial over BLE does work, but very unstable. It seems that it's affected by the network environment, signal strength, etc.

I would recommend to use USB when updating keymap for now

thomasaarholt, 2024-12-13:
That makes sense! 

I hope you don’t mind me making all these issues! I am REALLY enjoying working with rmk, and I am so satisfied with making my own keyboard.

closed 2024-12-13 by thomasaarholt

HaoboGu, 2024-12-13:
Not at all! I'm happy to hear feedback from users, whether it’s positive or negative :D

thomasaarholt, 2024-12-15:
I managed to get this [branch/PR of vial](https://github.com/vial-kb/vial-gui/pull/238) working (with quite a bit of trouble), and it works well with bluetooth :)

HaoboGu, 2024-12-15:
Interesting, I don't see anything changed that would fix the BLE connection failure. Any ideas?

thomasaarholt, 2024-12-18:
No. I was thinking initially that updating vial from Python 3.6 to anything else was bound to bring a lot of under-the-hood improvements, but there wasn't really anything standing out massively. But I haven't tested a lot.