rmk-rs/rmk #183: Pointing device support

HaoboGu, 2024-12-13:


HaoboGu, 2024-12-13:
Related issue: #29 #69

HaoboGu, 2024-12-24:
Since the generic task limitation(https://github.com/embassy-rs/embassy/issues/2454), the only way for defining a generic input device task is to use macro. There are two options:

1. Use a proc-macro like:
  ```rust
  #[input_device]
  struct RotaryEncoder {}
  impl InputDevice for RotaryEncoder {...}
  ```
  which could be expanded as:
  ```rust
  struct RotaryEncoder {}
  impl InputDevice for RotaryEncoder {...}
  #[embassy_executor::task]
  async fn rotaryencoder_task(device: RotaryEncoder) -> ! {
    device.run().await
  }
  ```
2. Use a decl macro
  ```rust
  struct RotaryEncoder {}
  impl InputDevice for RotaryEncoder {...}
  impl_input_device!(RotaryEncoder, task_name);
  ```
  
  expanded code:
  ```rust
  struct RotaryEncoder {}
  impl InputDevice for RotaryEncoder {...}
  #[embassy_executor::task]
  async fn task_name(device: RotaryEncoder) -> ! {
    device.run().await
  }
  ```

- option 1: simple and easy, but the task name cannot be customized
- option 2: the task name can be customized, but need call `impl_input_device` for each device

HaoboGu, 2024-12-24:
Okay, after some testing I found neither work with generic tasks. So I have to give up embassy task. The only option seems to be join all input device's `run` together:

```rust
// Implement the input device for RotaryEncoder
struct RotaryEncoder {}
impl InputDevice for RotaryEncoder {...}
```

in user space:

```rust
let e1 = RotaryEncoder {};
let e2 = RotaryEncoder {};
join(run_rmk(), run_devices!(e1, e2)).await
```

drindr, 2025-01-28:
I have an idea to make the input device/processor more flexible and avoid the unstable feature `#[feature(generic_const_exprs)]`

https://github.com/drindr/rmk/blob/3710603d4df71b5f27456617e48c41994b79effd/rmk/src/input_device/mod.rs#L103-L143

HaoboGu, 2025-01-28:
> I have an idea to make the input device/processor more flexible and avoid the unstable feature `#[feature(generic_const_exprs)]`
> 
> https://github.com/drindr/rmk/blob/3710603d4df71b5f27456617e48c41994b79effd/rmk/src/input_device/mod.rs#L103-L143

The main difference is to use a trait for receiver/sender instead of channels, am I right?

HaoboGu, 2025-02-26:
I'm thinking about removing `receiver`/`sender`, adding something like `DeviceBuilder` to bind the input device and corresponding processors. In current implementation, all the input device and processors are linked separately by channels, it's not easy for end users to configure them

HaoboGu, 2025-03-13:
End with something like:

```rust
run_devices!{
  (matrix, encoder, touchpad) => EVENT_CHANNEL
}

run_processor_chain! {
  EVENT_CHANNEL => [touchpad_processor, encoder_processor]
}
```

The channel can be customized, but I'm not sure that should we make channel customized? or just use built-in channels for all devices

drindr, 2025-03-13:
> End with something like:
> 
> run_devices!{
>   (matrix, encoder, touchpad) => EVENT_CHANNEL
> }
> 
> run_processor_chain! {
>   EVENT_CHANNEL => [touchpad_processor, encoder_processor]
> }
> 
> The channel can be customized, but I'm not sure that should we make channel customized? or just use built-in channels for all devices

I think it would be useful when a local processor is required, especially for split.
For instance, 
1. the boot/bootloader switch on the peripheral should be processed on the itself.  (a small keymap should be placed in the peripheral. Or just triggered by the master?)
2. the peripheral changes the master. (I'm not sure if someone have such a requirement)

HaoboGu, 2025-03-13:
> I think it would be useful when a local processor is required, especially for split. For instance,
>
> 1. the boot/bootloader switch on the peripheral should be processed on the itself.  (a small [...]

It runs on the peripheral side, we could have it as a proxy/filter of peripheral events, before the peripheral sends events out. Luckily it has nothing conflict with current implementation:D

HaoboGu, 2026-08-29:
Closing, the input device framework has been reworked, I'll keep working on polishing it.

closed 2026-08-29 by HaoboGu