rmk-rs/rmk #188: The keyboard cannot wake up the system from deep sleep.

Byvm815, 2024-12-15:
RMK Version: 0.4.4
RMK Feature: Split
Microcontroller: rp2040 4Mb
Platform: Linux

I am currently unable to determine whether the issue is caused by GNOME's power management or the BIOS deep sleep. I will provide additional details once I have identified the root cause of the issue.

thomasaarholt, 2024-12-15:
I think this PR does what you want, but you would need to use the main branch or wait for a new release.

https://github.com/HaoboGu/rmk/pull/184/files

Byvm815, 2024-12-15:
Thank you. When I was discussing with HaoboGu, he told me that the bug was fixed in the `feat/support_remote_wakeup` branch. However, I just couldn't find it in the repository.

thomasaarholt, 2024-12-15:
Hehe, branches are typically automatically deleted on merge, so that's not too surprising but not obvious! ☺️

HaoboGu, 2025-07-21:
Could you try latest version, it should be fixed in https://github.com/HaoboGu/rmk/commit/e8a49bc1d85b84c455248d3b8e9d54a01b3408d4

closed 2025-08-21 by HaoboGu

vladimiry, 2026-02-03:
@HaoboGu I just tested the case on Linux with recent RMK v0.8.2, using RP2040 and STM32F103VET6 - the system doesn’t wake from sleep when pressing the keyboard. It works as expected if I flash QMK to the same RP2040/STM32F103VET6-based boards. So I think this issue should be reopened.