rmk-rs/rmk #221: ModifierTapHold from keyboard.toml

hariseldon78, 2024-12-26:
Hello, thanks for this software. I am trying to implement the home row mods, and i would like to setup them in the keyboard.json. I think there is no way to do that at the moment, looking around in the code i only found the "LT(" parsing code.
Is there any plan of supporting it, and what will be the syntax? If it's already defined I could go on and try to post a PR myself.

HaoboGu, 2024-12-26:
Yes, ModifierTapHold can only be set by vial right now. Adding it should be easy, though.

hariseldon78, 2024-12-26:
ok. For adding it we could go 2 routes: supporting the already familiar syntax from QMK, which is
'''
LCTL_T(<keycode>)
LALT_T(<keycode>)
LSFT_T(<keycode>)
RCTL_T(<keycode>)
RALT_T(<keycode>)
RSFT_T(<keycode>)
'''

or a single pattern, like `MTH(<modifier>,<keycode>)`.

Which one would you prefer? (in my opinion the second one is cleaner, or we could do both)

hariseldon78, 2024-12-26:
ok, I'll just do both, and then we can remove the ones we don't like. I'll start tonight!

HaoboGu, 2024-12-26:
I prefer the second. It could be Mt(modifier_combination, key)

closed 2024-12-31 by hariseldon78