rmk-rs/rmk #225: Making more modules public

hyranno, 2025-01-01:
Requesting to make more modules public.
Hello and Happy New Year BTW.

### Context
I'm trying to make a keyboard with unconventional keyscan.
Then I wanted to write new struct implementing `matrix:MatrixTrait`, but `matrix` and relevant modules are not public.
So I request to make more modules public.

If you merge the codes for device into the rmk, making modules public is not necessary for it.
But making modules public is helpful when writing codes for minor devices which is not for merge.

Making lot of modules public may let new users confused.
If you want to emphasize commonly-used modules, how about making prelude module which re-exports commonly-used modules?

### Additional info
I'm working on my keyboard at https://github.com/hyranno/dflip_daisy .
I'm trying to make sequential keyscan using pins for column-clock, row-clock and reset.

HaoboGu, 2025-01-01:
Making `matrix` public makes sense. Could you please open a PR for it?

hyranno, 2025-01-01:
Can I take some time?
I haven't fully listed things to open to implement `MatrixTrait` and `run_rmk_...` using it.

hyranno, 2025-01-03:
Opened https://github.com/HaoboGu/rmk/pull/228
This should be enough to implement new `Matrix` and `run_rmk_...` outside rmk itself.

closed 2025-01-04 by hyranno