rmk-rs/rmk #229: Architecture upgrade, make input device - input processor - reporter decoupled with each other

HaoboGu, 2025-01-03:


hyranno, 2025-01-12:
Sharing idea.
Maybe it is nice to take `Sender` argument on input device (including `Matrix`) constructor, rather than designate in the input device.
Then input events will be able to be chained and transformed with like `Matrix` -> `SomeTransformer` -> `KEY_EVENT_CHANNEL`.
Here `SomeTransformer` transforms key events into another key events, can be like combo adapter, tap dance adapter, offset, merge, and debouncer.
`CentralMatrix` for split keyboard can be represented with offset and merge of sub matrices.
In this way `Matrix` can focus on reading signals.

HaoboGu, 2025-01-13:
> Maybe it is nice to take `Sender` argument on input device (including `Matrix`) constructor, rather than designate in the input device. Then input events will be able to be chained and transformed with like `Matrix` -> `SomeTransformer` -> `KEY_EVENT_CHANNEL`.

Does it mean that the input of an input device can be the output of another input device? Seems that #223 is a similar feature.

But personally, I have some concerns about it, because that requires users to organize their own input chain. The chain might become very long and be soon out of the control, especially for users who don't have experience on it.

hyranno, 2025-01-14:
Yes, that is what I mean.

Perhaps #223 can be done with vial, not sure though.
It looks like mapping of physical layout to logical row-column layout.

For longer common chain, wrapper or builder may help to encapsulate it?

HaoboGu, 2025-01-14:
> For longer common chain, wrapper or builder may help to encapsulate it?

I can't imagine how it could be simple and easy to use 😅

Having an example API design could be better for us to discuss about it

hyranno, 2025-01-15:
Well, I mean like this.
Considering commonly used device chain like below.
```
let device1 = Device1::new(KEY_EVENT_CHANELL);
let device2 = Device2::new(device1.event_sender());
let device3 = Device3::new(device2.event_sender());
```
It may be easier when tuple of devices can be treated as device.
```
impl<T1, T2> InputDevice for (T1, T2)
  where T1: InputDevice, T2: InputDevice {...}
impl<T1, T2, T3> InputDevice for (T1, T2, T3)
  where T1: InputDevice, T2: InputDevice, T3: InputDevice {...}
...
```
And provide builder for commonly used chain.
```
#[derive(Default)]
struct DefaultDeviceChainBuilder {
  opt1: Device1Option,
  opt2: Device2Option,
  opt3: Device3Option,
}
impl DefaultDeviceChainBuilder {
  pub fn set_opt1(self, opt: Device1Option) -> DefaultDeviceChainBuilder {...}
  pub fn set_opt2(self, opt: Device2Option) -> DefaultDeviceChainBuilder {...}
  pub fn set_opt3(self, opt: Device3Option) -> DefaultDeviceChainBuilder {...}
  pub fn build(self) -> Box<dyn InputDevice> {
    let device1 = match self.opt1 {
      Device1Option::Default => Device1::new(KEY_EVENT_CHANELL),
      Device1Option::Alt => Device1Alt::new(KEY_EVENT_CHANELL),
    };
    let device2 = match self.opt2 {
      Device2Option::Default => Device2::new(device1.event_sender()),
      Device2Option::Alt => Device2Alt::new(device1.event_sender()),
    };
    let device3 = match self.opt3 {
      Device3Option::Default => Device3::new(device2.event_sender()),
      Device3Option::Alt => Device3Alt::new(device2.event_sender()),
    };
    Box::new((device1, device2, device3))
  }
}
```
Then users can use it like below.
```
let device = DefaultDeviceChainBuilder::default().build();
```
or with some customization.
```
let device = DefaultDeviceChainBuilder::default().set_opt2(Device2Option::Alt).build();
```
In this way users does not need to know the whole chain.
And device features will stay small and reusable I think.

HaoboGu, 2025-01-16:
> #[derive(Default)]
> struct DefaultDeviceChainBuilder {
>   opt1: Device1Option,
>   opt2: Device2Option,
>   opt3: Device3Option,
> }

I have some questions about it:

1. The number of input devices seems limited in this chain builder, and we have to add repeated code for different device numbers(which might be solved by macros?), can it support any number of input devices?
2. If the user defines a ChainBuilder with 2 input devices, but fill only one of them, then the remaining devices in the chain is default ones, is it flash/ram usage friendly still?
3. I notice that there's `Box<dyn ..>` used. In no_std, it requires alloc, but now I prefer to not add any allocator to RMK. Is there any other ways to do the similar thing? I know `enum_dispatch`, but I'm afraid that it's impossible to use it for users' customized input devices. Do you know any other solutions for it?

hyranno, 2025-01-16:
Umm, without `Box<dyn T>` it looks like harder.

1. Using `Vec` with `impl InputDevice for Vec<Box<dyn InputDevice>>` may help for variable length, but it depends on `Box<dyn T>`.
2. Relying on `Box`, it may require heap size of maximum possible chain size. So it is not ram friendly.
3. Assuming device chain does not change after flush, theoretically the device chain and its size is known at compile time. So the builder can be implemented as macro, without `Box<dyn T>`.  It may return variable length of tuple and requires memory only actually needed. But honestly it is out of my knowledge.

HaoboGu, 2025-01-16:
Yeah, it's quite hard.. Anyway, thanks for sharing your ideas! I'll keep it in my mind to see if there better solutions for it.

PR #250 (2025-02-25): Project architecture upgrade
Closes #229 

The current impl adds 3 layers of the event processing pipeline: input devices, input processors and hid writers.

The input device reads physical inputs and generates the raw input events, the input processor processes and converts the raw events to hid reports, finally the hid writer writes the hid reports to the host. All 3 layers are event driven, which is compatible with embassy's async framework.
closed 2025-02-25 by HaoboGu