rmk-rs/rmk #231: Provide easy-to-use configuration for input devices in `keyboard.toml`

HaoboGu, 2025-01-03:


tib888, 2025-03-05:
See also https://github.com/HaoboGu/rmk/issues/268#issuecomment-2701991941 as an idea.

adong660, 2025-12-14:
> > Hi. I notice that the [docs](https://rmk.rs/docs/configuration/input_device/encoder.html#toml-configuration) documented on how to configure actions of rotary encoders by writing some encoder_map = .... But if I'm not mistaken, the current implementation sets all of them to k!(No) in the expanded function get_default_encoder_map. Are we going to implement it or seeking a more general way to configure encoders first?
> 
> [#669](https://github.com/HaoboGu/rmk/pull/669) has been merged, so the encoder configuration should work.
> 
> Edit: oh that part of doc should not appear in v0.8 doc, it might be a mistake. Anyway, the new version will be released soon

How about we put `encoder_map` under `[[layers]]` rather than under`[layout]`? Doing so, `[layout]` would become a description purely for the physical layout, and `[[layout]]` would handle all the key bindings. (After `layout.keymap` is completely deprecated.) It'd be clearer to recognize which encoder action corresponds to which layer, and be also easier for users to change the order of layers. An example would be:

```toml
[layout]
rows = 5
cols = 4
layers = 3
matrix_map = """
(0,0) (0,1)
"""

[[layers]]
name = "base_layer"
keys = """
A B
"""
encoders = [["C", "D"], ["E", "F"]]

[[layers]]
name = "number_layer"
keys = "..."
encoders = [["1", "2"], ["3", "4"]]
```