rmk-rs/rmk #247: RGB support

HaoboGu, 2025-01-14:


paul-fornage, 2025-08-22:
The only thing stopping me from making a keyboard is this feature support.

1. Is there a keyboard I can buy pre-made that supports this? I want to try out the project but not do any soldering until it's my own design
2. Any advice, considerations, ideas before I go crazy with mediocre PRs?

I honestly don't think I have the skill to make something that holds up to the standards of a repo this big, so I was going to make it local only, as in the host computer does not know about the RGB and either just use a couple basic presets mixed with some serial chain RGB chips (like WS2812), or make some basic integration with [WLED](https://github.com/WLED/WLED) on a separate esp32 so I can at least get some basic RGB functionality working for zoomers like me.

TLDR; I'm gonna do this for me, and if theres something I can do to make it fit for the project, tell me.

HaoboGu, 2025-08-22:
> Is there a keyboard I can buy pre-made that supports this? I want to try out the project but not do any soldering until it's my own design

I think any board that uses rp2040 is a good choice, or any zmk keyboard which has rgb support.

> Any advice, considerations, ideas before I go crazy with mediocre PRs?

The RGB support can be implemented using the [`Controller`](https://github.com/HaoboGu/rmk/blob/8f5f5213f1c0455f0d947a5932c0474e71a8c116/rmk/src/controller/mod.rs#L13), which is the standard way in RMK. You can take a look of controllers that in the community, like https://github.com/drindr/rmk-display

Also there's some libraries like WLED in Rust world, for example, [blinsky](https://github.com/ahdinosaur/blinksy).

jpds, 2026-10-22:
This merge and [v0.2.0](https://github.com/jpds/rmk-palettefx/commit/948ee1cb11390d2e00d24a358838d864cb542f16) were done a month ago.