rmk-rs/rmk #263: Enable Reset-Double-Tap for BOOTSEL for RP2040

atamanroman, 2025-02-20:
[QMK does](https://docs.qmk.fm/platformdev_rp2040#double-tap)

```
#define RP2040_BOOTLOADER_DOUBLE_TAP_RESET // Activates the double-tap behavior
#define RP2040_BOOTLOADER_DOUBLE_TAP_RESET_TIMEOUT 200U // Timeout window in ms in which the double tap can occur.
#define RP2040_BOOTLOADER_DOUBLE_TAP_RESET_LED GP17 // Specify a optional status led by GPIO number which blinks when entering the bootloader
```

for RP2040 so that double tapping RESET reboots in BOOTSEL.

Flashing RMK disables that feature, which "locked me out" for some time (I know, part of the game).

I'm not familiar enough with the codebase provide a PR, but I guess that would be an useful feature.

HaoboGu, 2025-02-23:
Yes, that's planned: #253

HaoboGu, 2025-03-14:
@atamanroman would you like to try #280? 

If `rp2040_bl` feature is enabled, you can jump to bootloader using vial

[image]

PR #280 (2025-03-14): Jump to bootloader key for rp2040
Closes #263