rmk-rs/rmk #272: Option to disable Vial configuration storage

artogahr, 2025-03-03:
A configuration option should exist that lets users configure their keymaps only via the keyboard.toml configuration. Ideally Vial would still be available for configuration testing, but the configuration would get reset on reboot. This would be different than `clear_storage = true` in that only the keymaps would get reset, not other irrelevant storage such as bluetooth configuration etc.

something like `disable_vial = true` or alternatively `use_vial = false`, or maybe even `save_vial_layout = false`

artogahr, 2025-03-04:
Wow, #257 is amazing and is a valid solution to this issue, will close it

HaoboGu, 2025-03-04:
> I think the `clear_storage` is unnecessary now in the main branch (unreleased yet) since [#260](https://github.com/HaoboGu/rmk/pull/260) has been merged. The experience about the keymap configuration will be nicer in the next version.😃

Actually it's not fully finished. #260 resets the whole storage -- which could be improved by setting only keymap. The BLE storage could be retained. If that, the `clear_storage` can be kept to reset storage(not only the keymap).

artogahr, 2025-03-04:
However I'm also thinking it might be a good idea to give users the option to _not_ clear the storage after an uodated firmware flash, will take a look at implementing that later.

Or not...? Idk I haven't checked the fill thing yet, possibly I'm missing some context / details

artogahr, 2025-03-04:
or nvm, reopening as just saw @HaoboGu 's comment :D

artogahr, 2025-03-04:
@HaoboGu if there's no urgency, can you assign this issue to me? I'd like to give a shot at implementing it.

HaoboGu, 2025-03-04:
Sure :D

artogahr, 2025-03-14:
Something like that is the original intention behind this, yes @gabevenberg

alex-tdrn, 2025-08-17:
I just saw this issue now. There is a rebased PR for a dedicated vial feature here https://github.com/HaoboGu/rmk/pull/520. I hadn't seen this issue when I started working on the PR, so I didn't know it was assigned to someone else, my bad.

The feature works, but the code is now littered with `#[cfg(feature = "vial")]` and is way harder to read in places (lib.rs is the worst file right now). I'm not sure how to properly decouple vial more from e.g. `run_rmk` and `run_keyboard`.

I will not have much time over the coming weeks (until around the end of september) to bring this over the finish line. If anyone wants to pick that PR up, you can gladly take over ✌️

artogahr, 2025-08-17:
Hi @alex-tdrn 

Thanks for taking the initiative, and feel free to, a lot has changed and unfortunately I don't have time to work on this.