rmk-rs/rmk #278: Increasing COMBO_MAX_NUM

willpuckett, 2025-03-11:
Is it possible to increase COMBO_MAX_NUM in keyboard.toml? I'm attempting to implement a keymap like the one below, which needs some 40 ish combos.

Also, combos don't seem to fire when they include an MT() keys. Is this intentional? 

[image]

willpuckett, 2025-03-11:
Ah, got the combos on the MT keys... I just have to enter the MT in the combo def...

HaoboGu, 2025-03-12:
It's not possible now. I did some initial work on it: https://github.com/HaoboGu/rmk/tree/feat/config_file but it's not finished.

willpuckett, 2025-03-12:
I'll give it a whirl!

HaoboGu, 2025-06-19:
Now the combo_max_num can be configured by `keyboard.toml`, see: https://rmk.rs/docs/features/configuration/rmk_config.html#behavior-configuration

closed 2025-06-19 by HaoboGu