rmk-rs/rmk #279: Build error when using direct pin and split on an RP2040

whitelynx, 2025-03-13:
I've opened a branch with a new example, `rp2040_split_pio_direct_pin`, which is based on `rp2040_split_pio`, but adds changes from `nrf52840_ble_split_direct_pin` to support using direct pins instead of a normal matrix. This fails to build:

```
   Compiling rmk v0.5.2 (/home/whitelynx/Development/Other/rmk/rmk)
   Compiling rmk-rp2040 v0.2.0 (/home/whitelynx/Development/Other/rmk/examples/use_config/rp2040_split_pio_direct_pin)
error[E0107]: function takes 6 generic arguments but 5 generic arguments were supplied
  --> src/peripheral.rs:6:1
   |
6  | #[rmk_peripheral(id = 0)]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^
   | |
   | expected 6 generic arguments
   | supplied 5 generic arguments
   |
note: function defined here, with 6 generic parameters: `In`, `Out`, `S`, `ROW`, `COL`, `SIZE`
  --> /home/whitelynx/Development/Other/rmk/rmk/src/split/peripheral.rs:89:14
   |
89 | pub async fn run_rmk_split_peripheral_direct_pin<
   |              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
90 |     #[cfg(feature = "async_matrix")] In: Wait + InputPin,
91 |     #[cfg(not(feature = "async_matrix"))] In: InputPin,
   |                                           --
92 |     Out: OutputPin,
   |     ---
93 |     #[cfg(not(feature = "_nrf_ble"))] S: Write + Read,
   |                                       -
94 |     const ROW: usize,
   |     ----------------
95 |     const COL: usize,
   |     ----------------
96 |     const SIZE: usize,
   |     -----------------
   = note: this error originates in the attribute macro `rmk_peripheral` (in Nightly builds, run with -Z macro-backtrace for more info)
help: add missing generic argument
   |
6  | #[rmk_peripheral(id = 0)], SIZE
   |                          ++++++

error[E0277]: the trait bound `Spawner: embedded_io_async::Write` is not satisfied
  --> src/peripheral.rs:6:1
   |
6  | #[rmk_peripheral(id = 0)]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `embedded_io_async::Write` is not implemented for `Spawner`
   |
   = help: the following other types implement trait `embedded_io_async::Write`:
             &embassy_sync::pipe::Pipe<M, N>
             &mut T
             &mut [u8]
             BufferedUartTx<'d, T>
             PioUartTx<'_, PIO, SM>
             embassy_rp::uart::BufferedUart<'d, T>
             embassy_sync::pipe::Pipe<M, N>
             embassy_sync::pipe::Writer<'_, M, N>
             rmk::split::rp::uart::BufferedUart<'a, PIO>
note: required by a bound in `run_rmk_split_peripheral_direct_pin`
  --> /home/whitelynx/Development/Other/rmk/rmk/src/split/peripheral.rs:93:42
   |
89 | pub async fn run_rmk_split_peripheral_direct_pin<
   |              ----------------------------------- required by a bound in this function
...
93 |     #[cfg(not(feature = "_nrf_ble"))] S: Write + Read,
   |                                          ^^^^^ required by this bound in `run_rmk_split_peripheral_direct_pin`
   = note: this error originates in the attribute macro `rmk_peripheral` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0277]: the trait bound `Spawner: embedded_io_async::Read` is not satisfied
  --> src/peripheral.rs:6:1
   |
6  | #[rmk_peripheral(id = 0)]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^ the trait `embedded_io_async::Read` is not implemented for `Spawner`
   |
   = help: the following other types implement trait `embedded_io_async::Read`:
             &[u8]
             &embassy_sync::pipe::Pipe<M, N>
             &mut T
             BufferedUartRx<'d, T>
             PioUartRx<'_, PIO, SM>
             embassy_rp::uart::BufferedUart<'d, T>
             embassy_sync::pipe::Pipe<M, N>
             embassy_sync::pipe::Reader<'_, M, N>
             rmk::split::rp::uart::BufferedUart<'a, PIO>
note: required by a bound in `run_rmk_split_peripheral_direct_pin`
  --> /home/whitelynx/Development/Other/rmk/rmk/src/split/peripheral.rs:93:50
   |
89 | pub async fn run_rmk_split_peripheral_direct_pin<
   |              ----------------------------------- required by a bound in this function
...
93 |     #[cfg(not(feature = "_nrf_ble"))] S: Write + Read,
   |                                                  ^^^^ required by this bound in `run_rmk_split_peripheral_direct_pin`
   = note: this error originates in the attribute macro `rmk_peripheral` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0061]: this function takes 4 arguments but 3 arguments were supplied
  --> src/peripheral.rs:6:1
   |
6  | #[rmk_peripheral(id = 0)]
   | ^^^^^^^^^^^^^^^^^^^^^^^^^
   | |
   | expected `bool`, found `BufferedUart<'_, PIO0>`
   | argument #4 of type `Spawner` is missing
   |
   = note: expected type `bool`
            found struct `rmk::split::rp::uart::BufferedUart<'_, PIO0>`
note: function defined here
  --> /home/whitelynx/Development/Other/rmk/rmk/src/split/peripheral.rs:89:14
   |
89 | pub async fn run_rmk_split_peripheral_direct_pin<
   |              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   = note: this error originates in the attribute macro `rmk_peripheral` (in Nightly builds, run with -Z macro-backtrace for more info)
help: provide the argument
   |
6  | #[rmk_peripheral(id = 0)](#[rmk_peripheral(id = 0)], /* bool */, #[rmk_peripheral(id = 0)], /* Spawner */)
   |                          +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++

Some errors have detailed explanations: E0061, E0107, E0277.
For more information about an error, try `rustc --explain E0061`.
error: could not compile `rmk-rp2040` (bin "peripheral") due to 4 previous errors
warning: build failed, waiting for other jobs to finish...
```

The `nrf52840_ble_split_direct_pin` example builds fine, as does the `rp2040_split_pio` example.

HaoboGu, 2025-03-13:
Could you try `feat/device_api` branch? I think I've fix it in this branch

HaoboGu, 2025-03-13:
This branch is quite ready to merge, I have to update documentations before that

gabevenberg, 2025-03-13:
> Could you try `feat/device_api` branch? I think I've fix it in this branch

Thanks, that works. (Im the person who ran headfirst into this on discord.)

HaoboGu, 2025-03-13:
> Could you try `feat/device_api` branch? I think I've fix it in this branch

I've merged it to main branch

whitelynx, 2025-04-01:
Sounds like we can close this one. 👍

closed 2025-04-01 by whitelynx