rmk-rs/rmk #285: state morph

tib888, 2025-03-15:
I'm planning to implement similar functionality to the "mod-morph" of ZMK or to the "fork" of kanata.
(It is also similar to "Key Overrides" in QMK, which would have the benefit to have VIAL support already, but too complex and missing the "else branch").

To goal is to be able to behave differently based on some state of the keyboard, most typically: which modifiers are active, but it would be nice if the lock-led states, layer states could be also used as condition (or the detected OS type if that would be available).

The needed parameters:
- keycode which triggers this behavior
- the list of state parameters, which are the base of the fork 
  (like: LShift, LCtrl, LAlt, LGui, RShift, RCtrl, RAlt, RGui, CapsLock, ScrollLock, NumLock, ...)
- the KeyAction to emmit when none of the listed states are active
- the KeyAction to emmit when any of the listed states are active

The simplest example use case:
```
[behavior.forks]
forks = [
  { trigger = ''BackSpace", conditions = ["LShift", "RShift"], none_output = "BackSpace", any_output = "Delete"  } #use Shift+BackSpace as Delete key
]
```

A complex use case: when "User0" KeyAction arrives, emmit "á" or "Á" depending on the shift/capslock state (on a US keybord on Windows, handle the case when the numlock is not pressed):
```
[behavior.forks]
forks = [
  { trigger = ''User0", conditions = ["LShift", "RShift", "CapsLock"], none_output = "User1", any_output = "User2"  } #capitalized letter decision
  { trigger = ''User1", conditions = ["NumLock"], none_output = "User3", any_output = "User4"  } #numlock missing decision
  { trigger = ''User2", conditions = ["NumLock"], none_output = "User5", any_output = "User6"  } #numlock missing decision
]

macros = [
  #on windows on the numpad type Alt+160 to emmit "á":
  { trigger ="User3",  [tap "NumLock", press "LAlt", tap "Kp1", tap "Kp6", tap "Kp0", release "LAlt", tap NumLock] },
  { trigger ="User4",  [press "LAlt", tap "Kp1", tap "Kp6", tap "Kp0", release "LAlt"] },
  #on windows on the numpad type Alt+181 to emmit "Á":
  { trigger ="User5",  [tap "NumLock", press "LAlt", tap "Kp1", tap "Kp8", tap "Kp1", release "LAlt", tap NumLock] },
  { trigger ="User6",  [press "LAlt", tap "Kp1", tap "Kp8", tap "Kp1", release "LAlt"] },
]
```
Unfortunately currently the macro configuaration possibility is missing for this later one so it is just "a pseudo config".
Using immediate "UserX" custom KeyActions is a trick to chain more than one forks depening on different state conditions.

The triggering KeyAction may come from Combos too.
The implementation must take care of the one-shot modifiers too.

So finding the right spot to process the configured forks will be tricky.

Any comments by anyone?

HaoboGu, 2025-03-16:
Consider the following config:

```
[behavior.forks]
forks = [
  { trigger = ''User0", conditions = ["LShift", "RShift", "CapsLock"], none_output = "User1", any_output = "User2"  } #capitalized letter decision
]
```

What will happen if I have a combo: `LShift` + `User0` = `User3`?

HaoboGu, 2025-03-16:
This feature could be very useful, I really want it

tib888, 2025-03-16:
> This feature could be very useful, I really want it

Well, it is almost ready with configuration, serialization, documentation, etc.
Only the most important part, the processing is far from perfect.

I tried these simple examples first:
```toml
forks = [
  # Shift + , output ; key
  { trigger = "Comma", none_output = "Comma", any_output = "Semicolon", conditions = ["LShift", "RShift"] },
  # Shift + . output : key
  { trigger = "Dot", none_output = "Dot", any_output = "WM(Semicolon, LShift)", conditions = ["LShift", "RShift"] },
]
```
The problems are:
- the . -> : works well, because the shift is already pressed and ":" needs the shift, but , -> ; also outputs : because the shifted state is not cleared while emmitting the replaced new keycode
- it should work with OSM, when evaluating the conditions, but the OSM should not affect the emmitted key afterwards (due to the previous point).
- if we want to allow chaining more than one of these forks then either we have to send the replacement key for reprocessing or the fork processor should run in a loop to find possible continuations (bad config may lead to infinite loops - needs protection!)
- the trigger key likely need to be limited to 'single keycode' and likely the output too if we want to eliminate the full reprocessing

To solve the first two points the keyboard should somehow enter into a temporary state with reconfigured modifiers, canceled OSMs, etc, then after the execution of the replaced keyaction the original state has to be restored (likely without the OSMs).
To control, which modifiers should be cancelled maybe more config parameters have to be exposed to the user (See: https://docs.qmk.fm/features/key_overrides has a lot of parameters).

Likely the temporary state problem can be solved with a little stack for previous KeyboardReport states.

Currently the most important question is: where should I insert the processing code.
I chosen this spot:

```rust
        if self.combo_on {
            if let Some(key_action) = self.process_combo(key_action, key_event).await {
                self.process_key_action(self.process_forks(key_action), key_event).await;
            }
        } else {
            self.process_key_action(self.process_forks(key_action), key_event).await;
        }
```

```rust
    fn process_forks(&self, key_action: KeyAction) -> KeyAction {
        if self.keymap.borrow().forks.len() > 0 {        
            for fork in &self.keymap.borrow().forks {
                if fork.trigger == key_action {
                    for condition in &fork.conditions {
                        if let KeyAction::Single(Action::Key(key)) = condition {
                            let any = match key {
                                KeyCode::LCtrl  => self.report.modifier & HidModifierBit::LCtrl as u8 != 0,
                                KeyCode::LShift => self.report.modifier & HidModifierBit::LShift as u8 != 0,
                                KeyCode::LAlt => self.report.modifier & HidModifierBit::LAlt as u8 != 0,
                                KeyCode::LGui => self.report.modifier & HidModifierBit::LGui as u8 != 0,
                                KeyCode::RCtrl  => self.report.modifier & HidModifierBit::RCtrl as u8 != 0,
                                KeyCode::RShift => self.report.modifier & HidModifierBit::RShift as u8 != 0,
                                KeyCode::RAlt => self.report.modifier & HidModifierBit::RAlt as u8 != 0,
                                KeyCode::RGui => self.report.modifier & HidModifierBit::RGui as u8 != 0,
                                KeyCode::NumLock => self.report.leds & 0x01 != 0,
                                KeyCode::CapsLock => self.report.leds & 0x02 != 0,
                                KeyCode::ScrollLock => self.report.leds & 0x04 != 0,
                                //KeyCode::Compose => self.report.leds & 0x08 != 0,
                                //KeyCode::Kana => self.report.leds & 0x10 != 0,
                                _ => false
                            };

                            if any {
                                return fork.any_output; 
                            }               
                        }
                    }                    
                    return fork.none_output;
                }
            }
        }
        key_action
    }
```

Do you have better idea?

tib888, 2025-03-16:
And one more serious problem:
this replacement has to happen both at the trigger key press and the trigger key release, but if the modifier released earlier than the trigger key, then we will see that the replacement key is pressed and an other one is released, so one key remains pressed in the buffer...

It would be simpler if this would be a conditional macro executor only, but that limits the possibilities.

tib888, 2025-03-16:
A more general problem, maybe a separate issue:
We have tools to add modifiers, but there is no way to suppres them. 
For example if i have a macro to type "hello" and then if I press the shift while taping M0, it will type "HELLO", which may be ok (?), but Ctrl+M0 will type Ctrl+H, Ctrl+E, Ctrl+L, Ctrl+L, Ctrl+O, what is not wanted...
(also the current macro implementation will not type "Hi" because WM(H, LShift) tap will not be executed...)

HaoboGu, 2025-03-17:
For macros, I prefer to cache the keyboard's state before executing macro and restore the state after macro. Similarly, modifier's state can be cached for key overriding feature. But you definitely don't want the recursion..

closed 2025-04-04 by tib888