rmk-rs/rmk #288: Tap-Hold early trigger during tapping term for HRM

xofyarg, 2025-03-17:
I currently have two issues when using RMK with HRM, both of them related to fast typing during the tapping term.

- Rolling keys are not registered as tapping. The sequence is: press A(a Tap-Hold key) -> press B -> release A -> release B. If allowed(maybe under an option?), this can be a shortcut to register an early TAPPING for A. Luckily it has been addressed by https://github.com/HaoboGu/rmk/pull/276/commits/d2d367bb21ede9b6666a37edacbef9acaf8fc5a1.

- Permissive hold doesn't work when connecting two Tap-Hold keys. E.g. press A(a Tap-Hold key) -> press B(another Tap-Hold key) -> release B -> release A. Again, during A's tapping term, because we already tapped B, A can be seen as HOLDING. The problem is, when processing A's press event, B's press and release event are pushed to the `unprocessed_events` queue, then A starts its hold action, so far so good. After that, B's press event is popped out from the queue, it will wait for events from the `KEY_EVENT_CHANNEL` source, until the tapping term ends. But B's release event is is in the `unprocessed_events` queue instead of `KEY_EVENT_CHANNEL` source, so B would never register as a TAPPING. I'm new to the codebase, not sure if we should have a way to peek the event source, or merge the unprocessed queue with the event source.

HaoboGu, 2025-03-17:
Thanks for proposing it!

The 1st issue is because that we don't have something like `hold-trigger-key-positions` in zmk, so if the first key is tap-hold key, it won't trigger `key streak` so the rolling fails. Hopefully #276 could solve it.

As for the second issue, I think `unprocessed_events.first()` could be used for peeking the next item in the queue. Any other ideas?

xofyarg, 2025-03-17:
> Hopefully https://github.com/HaoboGu/rmk/pull/276 could solve it.

I have a similar implementation locally, and it works pretty good so far.

> `unprocessed_events.first()` could be used for peeking the next item in the queue.

Not sure if I understood correctly, during a Tap-Hold key's tapping term, it waits for either the tap_hold timeout, or a event from the `KEY_EVENT_CHANNEL` async channel. Some of the events are read into the `unprocessed_events` queue, which is synchronized.

https://github.com/HaoboGu/rmk/blob/e16466d49f112f70212af2404426e91b3abf87c8/rmk/src/keyboard.rs#L559

So basically the events are in two separated sources, and the function `process_key_action_tap_hold` only read from one of them.

HaoboGu, 2025-04-01:
> So basically the events are in two separated sources, and the function process_key_action_tap_hold only read from one of them.

all events that received from `KEY_EVENT_CHANNEL` during a tap-hold key processing are pushed into `unprocessed_events`, they will be processed later after current tap-hold key is processed. This could be improved by caching all events(including the pressed event) and processing the events when detecting a release.

HaoboGu, 2025-07-04:
Fixed by #371

closed 2025-07-04 by HaoboGu