rmk-rs/rmk #296: Issues with base layer setting

Iaphetes, 2025-03-24:
Hi,

First of all I really enjoy the firmware you've been building. :)

It seems I've encountered an issue with the implementation of the DF and/or TG implementations.

# Expected behaviour
When using DF and or TG I would expect to switch to e.g. layer 4 to be able to still switch to lower layers using e.g. MO(1).

# Actual behaviour
However it seems that all these act (at least for me) like TO, disabling all other layers until I reset the keyboard to DF(0) using this modifier.

Maybe I'm misunderstanding the layer switches, but I'd need the described behaviour (just changing the base layer and still being able to access all other layers).

# My setup
- Dactyl Manuform Skeleton Edition 4x5
- KB2040s

HaoboGu, 2025-03-25:
If you change default layer to layer4, you could not access layer 1 until layer 4 is disabled

Iaphetes, 2025-03-25:
Ah okay. Thanks for the clarification.:)

Maybe this description could then make it into the documentation, so a user doesn't have to go hunting through QMK documentation.

closed 2025-03-25 by Iaphetes

HaoboGu, 2025-03-25:
Yes, 

I'll keep it open until documentation is added