rmk-rs/rmk #302: Combo issue

HaoboGu, 2025-03-27:
1. When there's overlapped combo configuration, such as `{ actions = ["J", "K"], output = "A" }` and `{ actions = ["J", "K", "L"], output = "B" }`, and processing `JKL`, the expected output should be `B`, but now `A` also can be triggered.
2. When using combinations of a combo like `{ actions = ["J", "K"], output = "LShift" }` and other keys, the `LShift` won't be triggered in some cases.

fxliang, 2025-03-28:
```toml
[behavior.combo]
timeout = "80ms"
combos = [
  {actions = ["F", "J"], output = "Enter"},
  {actions = ["S", "D"], output = "LShift"},
  {actions = ["K", "L"], output = "RShift"},
  {actions = ["E", "R"], output = "WM(LeftBracket, LShift)"},
  {actions = ["D", "F"], output = "WM(Kc9, LShift)"},
  {actions = ["C", "V"], output = "LeftBracket"},
  {actions = ["U", "I"], output = "WM(RightBracket, LShift)"},
  {actions = ["J", "K"], output = "WM(Kc0, LShift)"},
  {actions = ["M", "Comma"], output = "RightBracket"},
  {actions = ["A", "S", "D", "F"], output = "CapsLock"}
]
```

example, `ASDF` as `CapsLock`, when `SD` as `LShift`, `ASDF` sometime might be interrupted by `SD`

hold `SD` won't trigger a `LShift` holding

hitsmaxft, 2025-03-28:
https://github.com/hitsmaxft/rmk/commit/b24d83b905df2421beebf0f860d1054ffe1f4241

pcasotti, 2025-03-28:
> [@pcasotti](https://github.com/pcasotti) Could you please have a look?

Sure, I'll take a look later today

closed 2025-04-08 by HaoboGu